commaai/openpilot #25529: From "Camera malfunction" and subsequently "no panda", now my openpilot does not start

wongjunjin, 2022-08-23:
### Describe the bug

1. While driving, suddenly my openpilot has an error "Camera Malfunction" numerous times. 
2. Fixed the "Camera Malfunction" error by restarting comma and my car.
3. This happened a few times randomly in a span of 2 days (while driving).
4. Finally, when I restart my car, "no panda" appears and openpilot does not start anymore. 
5. I have reformat my OS and reinstall openpilot.
6. Still no difference. Having "no panda". 

### Provide a route where the issue occurs

7bc774bc8e17fd8a|2022-08-22--06-26-30

### openpilot version

0.8.13.1

adeebshihadeh, 2022-08-23:
This is unlikely a software bug, and we no longer support the comma two.

closed 2022-08-23 by adeebshihadeh

wongjunjin, 2022-08-24:
> To begin with, have you tried a different USB Type C cable? (USB 3.2 gen 2)

This is a good idea. let me try to look for another cable to see if it works. Will update