commaai/openpilot #25538: laikad: manage download cache

adeebshihadeh, 2022-08-24:
Laikad caches file to disk but doesn't clean up after itself. This happens to be mostly ok right now because it writes to a tmpfs which doesn't persist across boots. Laika should provide provide a function that returns whether a given file is relevant.

gijskoning, 2022-08-29:
Do you mean the files downloaded in Laika or the parsed orbits/nav data cached in Laikad? The latter is overwritten when new files arrive so that shouldn't be a problem. 
I think all files downloaded in Laika can be removed safely after 48 hours. (I think the prediction files contain data not longer than 48 hours). And most could actually be removed directly after being parsed since Laikad has its own cache. (except for the  .[attempt files](https://github.com/commaai/laika/blob/c8bc1fa01be9f22592efb991ee52d3d965d21968/laika/downloader.py#L283) to avoid many download requests)

adeebshihadeh, 2023-02-03:
@gast04 I think what we want to do here is refactor laika to not write to disk at all in our case, but it doesn't seem like that refactor will make it for 0.9.1. What do you think about just periodically cleaning the cache in laikad? Do we ever read it?

adeebshihadeh, 2023-02-03:
Ah, I see we're always running in no internet mode. Removing from the milestone.

haraschax, 2023-05-23:
Irrelevant until internet is enabled.

closed 2023-05-23 by haraschax