commaai/openpilot #25541: Navigation stopped when driving through a long tunnel

DerGue69, 2022-08-24:
### Describe the bug

I had navigation to a destination active.
Car entered tunnel. Navigation was awful then, GPS was all over the place.
When tunnel was over, I recognized that the navigation destination was cancelled.
It was not possible to reengage navigation through my smartphone.
[image]

### Provide a route where the issue occurs

Not recorded

### openpilot version

0.8.16 lfa-fix-warn

adeebshihadeh, 2022-08-26:
We can't look into this without a route. Why was this not recorded?

DerGue69, 2022-08-26:
I am from Austria.
I am currently in slowenia for a week.
Had to disable the upload to save data. (Personal sim card)

Will be back at home on Monday, try to provide route by then.

DerGue69, 2022-08-28:
6111bd7df5043394|2022-08-21--12-18-16--33   @   ~12:51 I entered the tunnel.
And afterwards the navigation was not working anymore.

sshane, 2023-08-08:
Does this still happen?

DerGue69, 2023-08-08:
I don't know, since I dont have prime any more.

closed 2024-06-20 by adeebshihadeh