commaai/openpilot #25569: Toyota: Prolonged low speed lateral oscillation

ntegan1, 2022-08-26:
### Describe the bug

This is an instance of steering oscillation commonly experienced when driving tssp corolla at low speeds (20->35mph).

https://user-images.githubusercontent.com/3820605/186997687-2c41ecb0-6f0d-449f-b786-21f99f4dc18d.mp4

(segment 9)
[image]

segments 10 and 11 also display oscillating torque output/steering angle.

### Which car does this affect?

Toyota Corolla 2017

### Provide a route where the issue occurs

a152dad3c3368320|2022-08-26--15-36-30--9

### openpilot version

38eb19bf04a5b6498cc38377796134205f0ee725

sshane, 2022-08-26:
Loving the use of the clipping feature!

ntegan1, 2022-08-31:
discord user sunny68 seems to be experiencing something similar (HKG i think)
https://1drv.ms/v/s!AunEjGlrWjRakCQ8T0fshxa3KtSn

#tuning https://discord.com/channels/469524606043160576/574796986822295569/1014285718073380926

ntegan1, 2022-09-02:
some drives on a152dad3c3368320|2022-09-01 with ~1/3 friction [code here](https://github.com/ntegan1/openpilot/tree/588660a82b46d71d1d7e975d5f6dba5b966a4471) seem to have less oscillation. feels smoother but doesn't want to turn as hard into sharper low speed curves as it needs to, so car ends up floating toward the double yellow lines/other lane and goes over occasionally (controller accuracy issue?).

```diff
-TOYOTA COROLLA 2017: [3.117154369115421, 1.8438132575043773, 0.12289685869250652]
+TOYOTA COROLLA 2017: [3.117154369115421, 1.8438132575043773, 0.04289685869250652]
```

Discord user Jordan suggests
```
jordan: In my 2010 Corolla, which has a similar steering setup to the 2017. I use MAX_LAT=3.6 FRICTION=0.032 and steer_actuator_delay=0.3875
```

will try out soon

adeebshihadeh, 2022-11-16:
Is this still an issue?

sshane, 2022-11-29:
@ntegan1 any updates?

closed 2022-11-29 by ntegan1

ntegan1, 2022-11-29:
> Is this still an issue?

Missed this. Much better!

Torque control is so good