commaai/openpilot #25578: Building openpilot fails on M1

MaherJendoubi, 2022-08-27:
### Describe the bug

maher@M1 openpilot %  pipenv shell                       
Loading .env environment variables...
Launching subshell in virtual environment...
Restored session: Sam 27 aoû 2022 18:57:01 CEST
 . /Users/maher/.local/share/virtualenvs/openpilot-eXNm2f47/bin/activate                                                                                      
maher@M1 openpilot %  . /Users/maher/.local/share/virtualenvs/openpilot-eXNm2f47/bin/activate
((openpilot) ) maher@M1 openpilot % scons -u -j$(nproc)                                                             
scons: Reading SConscript files ...
Git commit hash for gitversion.h: 04aeb30c
scons: done reading SConscript files.
scons: Building targets ...
gcc -o panda/tests/safety/test.os -c -nostdlib -fno-builtin -std=gnu11 -fPIC -Ipanda/tests/safety -Ipanda/board panda/tests/safety/test.c
panda/tests/safety/test.c:64:9: warning: 'NULL' macro redefined [-Wmacro-redefined]
#define NULL ((void*)0)
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/_types/_null.h:30:9: note: previous definition is here
#define NULL  __DARWIN_NULL
        ^
In file included from panda/tests/safety/test.c:66:
In file included from panda/board/safety.h:11:
panda/board/safety/safety_chrysler.h:82:19: error: initializer element is not a compile-time constant
  {CHRYSLER_ADDRS.CRUISE_BUTTONS, 0, 3},
   ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
panda/board/safety/safety_chrysler.h:88:26: error: initializer element is not a compile-time constant
  {CHRYSLER_RAM_DT_ADDRS.CRUISE_BUTTONS, 2, 3},
   ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
panda/board/safety/safety_chrysler.h:94:26: error: initializer element is not a compile-time constant
  {CHRYSLER_RAM_HD_ADDRS.CRUISE_BUTTONS, 2, 3},
   ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
panda/board/safety/safety_chrysler.h:100:28: error: initializer element is not a compile-time constant
  {.msg = {{CHRYSLER_ADDRS.EPS_2, 0, 8, .check_checksum = true, .max_counter = 15U, .expected_timestep = 10000U}, { 0 }, { 0 }}},
            ~~~~~~~~~~~~~~~^~~~~
panda/board/safety/safety_chrysler.h:110:35: error: initializer element is not a compile-time constant
  {.msg = {{CHRYSLER_RAM_DT_ADDRS.EPS_2, 0, 8, .check_checksum = true, .max_counter = 15U, .expected_timestep = 10000U}, { 0 }, { 0 }}},
            ~~~~~~~~~~~~~~~~~~~~~~^~~~~
panda/board/safety/safety_chrysler.h:119:35: error: initializer element is not a compile-time constant
  {.msg = {{CHRYSLER_RAM_HD_ADDRS.EPS_2, 0, 8, .check_checksum = true, .max_counter = 15U, .expected_timestep = 10000U}, { 0 }, { 0 }}},
            ~~~~~~~~~~~~~~~~~~~~~~^~~~~
1 warning and 6 errors generated.
clang -o selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/lat_cost/lat_cost_y_e_hess.os -c -std=gnu11 -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -DACADOS_WITH_QPOASES -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -Wno-unused -fPIC -I/opt/homebrew/include -I/opt/homebrew/opt/openssl/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/curl/include -Ithird_party/libgralloc/include -Ithird_party/android_frameworks_native/include -Ithird_party/android_hardware_libhardware/include -Ithird_party/android_system_core/include -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/lat_cost/lat_cost_y_e_hess.c
clang -o selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/lat_cost/lat_cost_y_0_fun.os -c -std=gnu11 -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -DACADOS_WITH_QPOASES -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -Wno-unused -fPIC -I/opt/homebrew/include -I/opt/homebrew/opt/openssl/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/curl/include -Ithird_party/libgralloc/include -Ithird_party/android_frameworks_native/include -Ithird_party/android_hardware_libhardware/include -Ithird_party/android_system_core/include -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/lat_cost/lat_cost_y_0_fun.c
clang -o selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/lat_cost/lat_cost_y_0_fun_jac_ut_xt.os -c -std=gnu11 -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -DACADOS_WITH_QPOASES -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -Wno-unused -fPIC -I/opt/homebrew/include -I/opt/homebrew/opt/openssl/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/curl/include -Ithird_party/libgralloc/include -Ithird_party/android_frameworks_native/include -Ithird_party/android_hardware_libhardware/include -Ithird_party/android_system_core/include -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/lat_cost/lat_cost_y_0_fun_jac_ut_xt.c
clang -o selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/lat_cost/lat_cost_y_0_hess.os -c -std=gnu11 -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -DACADOS_WITH_QPOASES -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -Wno-unused -fPIC -I/opt/homebrew/include -I/opt/homebrew/opt/openssl/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/curl/include -Ithird_party/libgralloc/include -Ithird_party/android_frameworks_native/include -Ithird_party/android_hardware_libhardware/include -Ithird_party/android_system_core/include -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/lat_cost/lat_cost_y_0_hess.c
cd /Users/maher/Documents/openpilot/selfdrive/controls/lib/longitudinal_mpc_lib && python3 long_mpc.py
clang++ -o selfdrive/locationd/generated/gps.o -c -std=c++1z -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -I/opt/homebrew/include -I/opt/homebrew/opt/openssl/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/curl/include -Ithird_party/libgralloc/include -Ithird_party/android_frameworks_native/include -Ithird_party/android_hardware_libhardware/include -Ithird_party/android_system_core/include [... cut ...]
 If there is an incompatibility with the CasADi generated code, please consider changing your CasADi version.
Version 3.5.4+ currently in use.
Warning: Please note that the following versions of CasADi  are officially supported: 3.5.5 or 3.5.4 or 3.5.3 or 3.5.2 or 3.5.1 or 3.4.5 or 3.4.0.
 If there is an incompatibility with the CasADi generated code, please consider changing your CasADi version.
Version 3.5.4+ currently in use.
scons: building terminated because of errors.

### OS Version

macOS Monterey Version 12.5.1

### openpilot version or commit

13489d092e2c09119e880022507d7331cf54c615

liangjiali0304, 2022-08-31:
Did you happen to solve this issue?

MaherJendoubi, 2022-08-31:
No, the issue remains.

adeebshihadeh, 2022-12-31:
Duplicate of #25112

closed 2022-12-31 by adeebshihadeh