commaai/openpilot #25586: Mazda: 2022 CX5 disengage on gas triggers stock radar cruise control cancelled message

alancyau, 2022-08-28:
### Describe the bug

I noticed a pattern where pressing the gas pedal while assisting the steering wheel during a EPS limit exceeded situation causes the car to display a "Mazda Radar Cruise Control Cancelled" message on the dash and a chime. OpenPilot will still properly disengage and this is simply a deviation from a regular disengage. A regular disengage on gas, not during EPS limit exceeded situation, will not produce this message.

a4af1602d8e668ac|2022-08-28--15-20-10--4

### Which car does this affect?

2022 Mazda CX5

### Provide a route where the issue occurs

a4af1602d8e668ac|2022-08-28--15-20-10--4

### openpilot version

082e46ce5e0f16108d59e8c2c04ec76ea8e7dd86

alancyau, 2022-08-28:
[image]

alancyau, 2022-08-28:
[image]

alancyau, 2022-09-02:
This bug appears to be resolved on @Jafaral's branch. https://github.com/Jafaral/openpilot/tree/mazda-tja

a4af1602d8e668ac|2022-09-02--22-30-21--10

sshane, 2023-08-08:
Do you know what he does?

Jafaral, 2023-08-08:
I think I resolved upstream too.

sshane, 2023-08-08:
What was the issue?

closed 2023-08-08 by sshane