commaai/openpilot #25593: 2020 GMC Sierra 1500

morrislee, 2022-08-29:
### Describe the bug

Sometimes braking manually to disengage Openpilot on 0.8.16 release3 will trigger error for "Cruise Faulted" and also makes the vehicle trigger forward collision alert. This happens very frequently when using brake to disengage Openpilot regardless if there is a lead car or not.

Route with logs uploaded

fe80a4e1cedec853|2022-08-28--11-41-21--4

### Which car does this affect?

GMC Sierra 1500 2020

### Provide a route where the issue occurs

fe80a4e1cedec853|2022-08-28--11-41-21--4

### openpilot version

0.8.16

morrislee, 2022-08-31:
Adding another route with this issue with logs uploaded. This happened at the end of the route

fe80a4e1cedec853|2022-08-30--12-24-25--5

https://connect.comma.ai/fe80a4e1cedec853/1661883865882/1661884212658

adeebshihadeh, 2022-08-31:
Thanks for the report (and uploading logs!), @sshane let's get this fixed for 0.8.17

sshane, 2022-08-31:
Talking with the user on Discord!

morrislee, 2022-10-17:
looks like shane performed his magic work and solved the problem! marking it closed now

closed 2022-10-17 by morrislee