commaai/openpilot #25603: proclog: missing processes sometimes

adeebshihadeh, 2022-08-29:
CI occasionally hits this running `test_onroad.py`, here's an example failure:

```
..dict_keys(['/sbin/init', '/lib/systemd/systemd-journald', '/lib/systemd/systemd-udevd', '/lib/systemd/systemd-networkd', '/lib/systemd/systemd-resolved', '/usr/lib/accountsservice/accounts-daemon', '/usr/bin/adsprpcd', '/usr/local/pyenv/versions/3.8.10/bin/python', '/usr/sbin/cron', '/usr/bin/dbus-daemon', '/usr/sbin/NetworkManager', '/sbin/leprop-service', '/usr/sbin/cnss_daemon', '/usr/bin/python3', '/usr/lib/policykit-1/polkitd', '/usr/sbin/rsyslogd', '/sbin/logd', '/usr/sbin/smartd', '/lib/systemd/systemd-logind', '/usr/bin/weston', '/sbin/wpa_supplicant', '/sbin/tftp_server', '/usr/sbin/atd', '/usr/sbin/ModemManager', '/usr/bin/cdsprpcd', 'sleep', '/usr/bin/tmux', '/usr/bin/bash', '/usr/sbin/chronyd', '/sbin/rmt_storage', '/sbin/init_mss', '/sbin/agetty', '/usr/bin/pdmappersvc', '/usr/libexec/weston-keyboard', '/usr/libexec/weston-desktop-shell', 'sshd: /usr/sbin/sshd -D [listener] 0 of 10-100 startups', '/usr/libexec/qmi-proxy', '/lib/systemd/systemd', '(sd-pam)', '/usr/bin/pulseaudio', 'sshd: comma [priv]', 'sshd: comma@pts/1', '/usr/local/pyenv/versions/3.8.10/bin/python3', 'python', './_ui', 'system.logmessaged', 'system.timezoned', './_soundd', 'selfdrive.loggerd.deleter', 'selfdrive.boardd.pandad', 'selfdrive.thermald.thermald', 'selfdrive.tombstoned', 'selfdrive.loggerd.uploader', 'selfdrive.statsd', 'selfdrive.athena.athenad', './boardd', './camerad', './clocksd', './logcatd', './proclogd', './_dmonitoringmodeld', './encoderd', './_modeld', './_sensord', './ubloxd', './locationd', 'selfdrive.locationd.calibrationd', 'selfdrive.controls.controlsd', 'selfdrive.monitoring.dmonitoringd', 'selfdrive.locationd.laikad', 'selfdrive.navd.navd', 'selfdrive.locationd.paramsd', 'selfdrive.sensord.pigeond', 'selfdrive.controls.plannerd', 'selfdrive.controls.radard'])

------------------------------------------------

------------------ CPU Usage -------------------

------------------------------------------------

selfdrive.controls.controlsd         37.15% (35.00%) 

./loggerd                             0.00% (10.00%) NO METRICS FOUND

./encoderd                           17.32% (17.00%) 

./camerad                            14.53% (14.50%) 

./locationd                           7.64% ( 9.10%) 

selfdrive.controls.plannerd          14.73% (11.70%) 

./_ui                                19.45% (19.20%) 

selfdrive.locationd.paramsd           8.62% ( 9.00%) 

./_sensord                            7.38% ( 6.17%) 

selfdrive.controls.radard             4.51% ( 4.50%) 

./_modeld                             6.28% ( 4.48%) 

./boardd                              4.01% ( 3.63%) 

./_dmonitoringmodeld                  5.05% ( 5.00%) 

selfdrive.thermald.thermald           3.85% ( 3.87%) 

selfdrive.locationd.calibrationd      2.19% ( 2.00%) 

./_soundd                             1.75% ( 1.00%) 

selfdrive.monitoring.dmonitoringd     3.87% ( 4.00%) 

./proclogd                            1.73% ( 1.54%) 

system.logmessaged                    0.24% ( 0.20%) 

./clocksd                             0.00% ( 0.02%) 

./ubloxd                              0.47% ( 0.02%) 

selfdrive.tombstoned                  0.02% ( 0.00%) 

./logcatd                             0.09% ( 0.00%) 

------------------------------------------------
```

adeebshihadeh, 2022-08-29:
@deanlee want to check this one out?

deanlee, 2022-08-30:
checking

deanlee, 2022-08-30:
Still debugging, but I suspect that `loggerd` has become a zombie process, /proc/[loggerd_pid]/cmdline is empty in this case.

adeebshihadeh, 2023-04-19:
This hasn't happened in a log time, closing for now.

closed 2023-04-19 by adeebshihadeh