commaai/openpilot #25731: Wi-Fi networks with WPA3 are not supported

TheOfficialJosh, 2022-09-12:
Wi-Fi networks using only the WPA3 security protocol are showing the 🚫 symbol next to them, and cannot be joined.

- The Qualcomm 845 does seem to support WPA3 at a hardware level. (Source: [XDA Developers](https://www.xda-developers.com/the-qualcomm-snapdragon-845-will-support-the-new-wpa3-wi-fi-security-standard/))
- Discord user nelsonjchen said the missing Security Flag might be "NM_802_11_AP_SEC_KEY_MGMT_SAE". ([Link to post](https://discord.com/channels/469524606043160576/871838269405556736/1018542403067584605))

See the network "DingoAteYourBaby" in the attached screenshot, which is only using WPA3. The IOT network current connected, has the exact same settings, on the same Access Point, except it is using WPA2.
[image]

nelsonjchen, 2022-09-12:
https://github.com/commaai/openpilot/blob/f6398ea7864779f621f84df65ded7ee083e72c09/selfdrive/ui/qt/offroad/wifiManager.cc#L141

Probably just needs an extra flag added here. 

I would also say this issue probably only applies to networks with WPA3 only.

adeebshihadeh, 2022-09-12:
Great, want to open a PR and test it?

TheOfficialJosh, 2022-09-12:
I am by no means a developer, but I did fork and add the flag in both networkmanager.h & wifiManager.cc.
https://github.com/TheOfficialJosh/openpilot/tree/wifi-wpa3

It sees the network and allows me to type in the password, but it does not actually connect.

TheOfficialJosh, 2022-09-12:
Tested with two different Wi-Fi networks using only WPA3. It only saves the password, but never connects. Going to guess there's something more than these two lines.

Is there a log file I can provide to help with figuring it out?

incognitojam, 2022-09-13:
I can replicate this on my comma three, and looking at the system logs doesn't give much information. I think we might just have to update the network manager package, which is already on the latest release available in apt for Ubuntu 20.04/agnos...

adeebshihadeh, 2022-09-13:
We [build ModemManager from source](https://github.com/commaai/agnos-builder/blob/master/userspace/compile-modemmanager.sh), might have to do the same for NetworkManager.