commaai/openpilot #25747: Make VIN query work on non-OBD bus

adeebshihadeh, 2022-09-12:


sshane, 2023-08-01:
- [ ] should find some other way to know when the `sendcan` socket is fully connected in boardd so we can remove all the retries we do. Really should successfully respond on the first try. Adding a second bus now will double the VIN query time to 2 seconds for cars that don't respond

sshane, 2024-02-01:
done! https://github.com/commaai/openpilot/pull/31165

the time changed from 1s to 1.5s with the recent changes, only if the VIN doesn't come back.

If it only responds on bus 1 it will take 0.4s-0.5s, and if it responds on bus 0, it will take 0.1s to 0.2s. Bolt always will take 0.3s. @adeebshihadeh are you fine with shipping this to release?

closed 2024-02-01 by sshane