commaai/openpilot #25751: Honda: query FW versions on non-OBD bus

sshane, 2022-09-12:


sshane, 2022-09-12:
@csouers want to take a stab at this/test on any cars you have access to?

sshane, 2022-11-30:
moved to https://github.com/commaai/openpilot/issues/26643

csouers, 2022-12-13:
Here's what I got from my own car. Some of the responses with the `--nonstandard` flag look to have what might be hardware part number data, but aside from that, nothing interesting like a response from a new module. Some part of the query process did cause faults to show on the headunit. I didn't investigate into the root cause of that. 

I did have to disable some checks (against unexpected responses) in the uds library in order to complete the query. I used the query script in panda/examples for this.

can 1 in normal mode:
https://gist.github.com/csouers/bb6abba0658609484f29572070190ca4

can 1 in obd2 mode:
https://gist.github.com/csouers/7dc1d0227c1024de6744c409d4213a6d

The private can (bus 0 and 2) between the camera and radar yielded no responses whatsoever.

closed 2024-03-21 by sshane