commaai/openpilot #25754: Subaru: query FW versions on non-OBD bus

sshane, 2022-09-12:


sshane, 2022-11-30:
moved to https://github.com/commaai/openpilot/issues/26643

jnewb1, 2023-11-29:
tried on the outback. bus 0 you only get the camera, bus 1 you get powertrain ecus (transmission, engine, abs). Couldn't get eps on any bus

sshane, 2023-11-29:
Great, let's add bus 1 (no OBD) and 0 then. Can you also check later with panda's query_fw_versions.py + `--nonstandard` with a good scan of each bus to see if there's another ECU we're not currently querying that we might want to? I saw 0x730 returns something, that's the adas ECU on Hyundai.

sshane, 2024-01-26:
Ah, we don't for gen 1:

```
               'carFw': [{'address': 1862,
                          'brand': 'subaru',
                          'bus': 1,
                          'ecu': 'eps',
                          'fwVersion': b'\n\xc0\x04\x00',
                          'logging': False,
                          'obdMultiplexing': True,
                          'request': [b'>\x00', b'"\xf1\x82'],
                          'responseAddress': 1870,
                          'subAddress': 0},
```

closed 2024-01-27 by jnewb1