commaai/openpilot #25758: Replace cabana with Qt app

adeebshihadeh, 2022-09-13:
[cabana](https://github.com/commaai/cabana), our web-based CAN reverse engineering tool, has become difficult to maintain, especially as we added CAN-FD. I think it's time to re-write cabana to make the quality more similar to the rest of our tools; most of the important components should be shared with openpilot: CANParser, camera view widget, etc.

We can build this very incrementally, like we did with replay. We should also get things like streaming from the device for free, since it'll just use all the normal openpilot stufd. @deanlee want to work on this? I'm thinking we use [PySide](https://wiki.qt.io/Qt_for_Python), but open to c++ if you have a strong preference.

deanlee, 2022-09-16:
WIP, I need some time to read the original code and fully understand it's logic.

JasonJShuler, 2022-10-08:
Will it be possible to ensure the solution is cross-platform? Cabana as-is is very glitchy, but it runs on Windows, Mac, even Android. qt is pretty Linux-centric. It is my understanding that it can be cross-platform (i.e. stock PlotJuggler runs on windows, but the OP plugins do not)

Just thinking about the broader audience - I'm 90% linux these days and the idea of a non-browser-based cabana is exciting. Also don't want to lose the contributors who aren't running linux.

JasonJShuler, 2022-10-10:
> ...  I'm thinking we use [PySide](https://wiki.qt.io/Qt_for_Python), but open to c++ if you have a strong preference.

FWIW, I second the preference for Python over c++

deanlee, 2022-10-10:
> FWIW, I second the preference for Python over c++

With C++, we can re-use many well-implemented libs such as opendbc, replay, util, camerawidth, to implement a high-performance cabana with only 1500 lines of code.
 
Using python will greatly increase the workload and complexity. (And I think there is a high probability that it will not run on the window too, if that's what you need.)

JasonJShuler, 2022-10-10:
> With C++, we can re-use many well-implemented libs such as opendbc, replay, util, camerawidth, to implement a high-performance cabana with only 1500 lines of code.
>
> Using python will greatly [...]

Oh ok; sounds like c++ makes more sense. I just personally have a lot of trouble understanding some c++ code.

Also, I deleted the comment about Windows support - I decided more motivation for people to use Linux is good for the universe :)

Thanks!

adeebshihadeh, 2022-10-12:
Congrats on the first release @deanlee!

closed 2022-10-12 by adeebshihadeh