commaai/openpilot #25835: Honda: Dashboard errors

alancyau, 2022-09-18:
### Describe the bug

When starting the car and within the first couple of seconds, the dashboard lights up with almost every error (brake system, TPMS, forward collision, etc) and the C3 displays `CAN Error: Check connections`. The car will still drive fine and will go away after driving a few miles and restarting the car. Noticed this happening the last couple of weeks. Usually once or twice a week but no obvious pattern. Since the errors would show up right away when the car started, I didn't think the C3 would be able to log anything. However, this time the dashboard errors started after about 20 seconds while the car was in drive, so I'm hoping something can be seen in the logs.

I've since removed the C3 but kept the harness and harness box installed to see if this is isolated to just the C3.

[image]
[image]

### Which car does this affect?

2017 Honda CRV EPS mod

### Provide a route where the issue occurs

a4af1602d8e668ac|2022-09-18--10-13-38--0

### openpilot version

0.8.16

sshane, 2022-09-19:
It looks like some messages (VSA_STATUS for example) time out after starting the car, perhaps try a different cable?

alancyau, 2022-09-19:
> It looks like some messages (VSA_STATUS for example) time out after starting the car, perhaps try a different cable?

Thanks for checking the logs. This is on the original C3 OBDC cable. I'll try another one.

mdajr, 2022-09-22:
I've also seen this multiple times since the 0.8.15 release. I reverted back to 0.8.14 and did not see the issue return for over a month. Upgraded to 0.8.16 - saw the issue almost immediately. Upgraded to master-ci (8d11e24) and saw the issue again.

Because it can't be reproduced on 8.14, I'm not inclined to think it's a hardware issue. If it were, maybe some timings were tightened that exposed the bad hardware?

Latest segment: f369374ecb30d9ad|2022-09-22--12-04-14--0

alancyau, 2022-09-26:
I switched to another OBDC cable and still got the same problem. I agree with @mdajr, this seems to be only happening in more recent versions. This didn't happen roughly 6 months ago. 

This only occurs during the initial minute of the vehicle starting. I've never experienced this error in the middle of a drive while OP is engaged, which makes me think that it's not a hardware issue.

sshane, 2023-01-17:
@alancyau are you seeing this issue recently at all? I doubt it's a software issue since my original diagnosis was seeing certain CAN messages stop being sent on the bus. It's likely either faulty car ECUs, or the cable or harness box is somehow dropping/blocking messages causing dash warnings.

alancyau, 2023-01-17:
Unfortunately I sold the car already. Maybe @mdajr can share his experiences. I have a friend with a 18 CRV that also experienced this issue. I'll check with him and report back. Both our CRVs had EPS modded, not sure if that matters.

Shouldn't be related to heat since it occurred during September time frame where the temperature is no higher than 70F here. Also the car would be parked in a covered garage.

mdajr, 2023-01-17:
Unfortunately I still see it on occasion when the C3 does a cold boot. I'm still sure on how it can be hardware since a rollback to 8.14 resolves the issue - but sshane obviously has a lot of knowledge of the code base (compared to my zero).

Can someone point me to documentation on capturing debug? Assuming there's more logging then what's uploaded (?) I can help pull some more thorough logs.

sannyd, 2023-05-20:
It only happens when the comma 3 is “shut down”. Perhaps this is because of battery protection? If it’s not shut down works fine with no errors. 

Another way to get around it is to put the car in run but not start until into the bot half way then start and no problem either.

adeebshihadeh, 2023-06-25:
@mdajr @sannyd can you guys provide a route where this happens? Just copy the link from connect and make sure to click the "upload logs" button and connect to wifi

mdajr, 2023-06-25:
I'm not sure if this is expired or not: f369374ecb30d9ad|2022-09-22--12-04-14--0

If it is, I'll post a new one when it happens next

adeebshihadeh, 2023-06-25:
Yeah, the logs are gone from that one. A route from the latest release or master is best.

mdajr, 2023-06-25:
Was able to reproduce it this morning. All I did was start the car, plug in the comma3, waited for it to boot up and -bam- Christmas tree

f369374ecb30d9ad|2023-06-25--10-13-13--0

sannyd, 2023-07-04:
a3fc26a4da829bb9|2023-07-04--10-16-36--0

sannyd, 2023-08-07:
a3fc26a4da829bb9|2023-08-06--19-10-55--0

denboozle, 2023-12-18:
I've had the same problem on my 2017 Honda CR-V. I started with the Comma 2 in April 2020. I upgraded from the Comma 2 to the Comma 3, hoping the problem would go away. I cancelled the connect plan because most of the time I don't have the unit connected. When I first got the Comma 2, the event was infrequent. Maybe a few months before I upgraded to the Comma 3, the event became frequent, and has always been frequent since.

Here are a few differences I have. Because I've been using this, on and off, for several years, I HAVE seen it fail in the middle of a long drive more than once. But, that is less frequent. Most of the time, the CAN Error happens at startup or within the first few minutes.

I don't have the EPS mod. I can leave the harness plugged in. I've replaced the cables and flipped the USB-C cable .

Here is what I'm trying now, because I *love* this thing when it works. I keep the unit in the house at room temperature. When I *really* want to use it, I'll plug it in after I've been driving a while, maybe 10 to 15 minutes after the drive has started. I leave it plugged in until completing the last drive of the day, then take it back into the house.
[image]

sshane, 2024-03-20:
Does this still happen? Can you provide recent routes if so?

denboozle, 2024-03-20:
That's a good question. Mostly, I connect the unit only when I take a long drive. For my last long drive of 12 hours, several weeks ago, I had no issues. Tomorrow, I'll connect the unit and leave it connected so that it will turn on and off for more short trips.

mdajr, 2024-03-28:
I have not seen the issue for a while now, and I remember seeing that there were recent fixes for race conditions related to Honda Bosch CAN addrs. We can close this for now and re-open if need be.

denboozle, 2024-03-28:
I have not seen the problem either, after leaving the unit connected for the last week. So, I agree we can close it. Thanks.

closed 2024-03-28 by adeebshihadeh