commaai/openpilot #25882: Honda CR-V Car Unrecongnized error

Torreskai0722, 2022-09-23:
### Describe the bug

I have used comma 3 on my Honda CR-V for severla months and it works well. Recently it shows error that the car is unrecognized. Sometimes I can resolve this error after unplug and plut the OBD adapter. Now it cannot work whatever I do.

### Which car does this affect?

Honda CR-V 2021

### Provide a route where the issue occurs

N/A

### openpilot version

0.9.16

### Additional info

[comma3.pdf](https://github.com/commaai/openpilot/files/9637207/comma3.pdf)

incognitojam, 2022-09-24:
Please provide your dongle ID so that it can be investigated

sshane, 2022-09-28:
Can you provide your dongle id or route name? We can't look into this issue without any data

sshane, 2022-10-01:
Any updates @Torreskai0722?

Torreskai0722, 2022-10-01:
> Any updates @Torreskai0722?

Thanks for the reply. The issue is not resolved. The dongle id is 64fb2e43102e56d5.

sshane, 2022-10-01:
It looks like your comma three isn't able to communicate with the comma power. I would try a new ethernet cable and comma power. Also try to reseat all connections

sshane, 2023-01-13:
@Torreskai0722 it looks like you haven't driven since October, so I'll close this issue for now in case you've returned the C3. Please let us know if you are still struggling with this issue and we will re-open!

closed 2023-01-13 by sshane

Torreskai0722, 2023-01-16:
The former issue disappear but another issue comes out. When the comma 3 device is powered, there are strange sounds like marble hitting. I have attached a video for it.

https://user-images.githubusercontent.com/36166171/212577319-ce0ca18e-6266-4d66-a6de-1d24ba15a3a9.mp4

sshane, 2023-01-16:
I would email support for that issue.

Torreskai0722, 2023-01-25:
> I would email support for that issue.

Any update on that?

sshane, 2023-01-25:
Did you email support?