commaai/openpilot #25884: Getting car unrecognized message on supported car

jlhsr1, 2022-09-24:
### Describe the bug

Getting car unrecognized message on supported car. Subaru Legacy 2021. Also message saying either unsupported or its ECUs are not recognized. New Comma 3 device and is working on my 2020 Subaru Ascent.  Thanks for any help. I'm 74 and this a little  over my head. John

### Which car does this affect?

2021 Subaru Legacy

### Provide a route where the issue occurs

no route

### openpilot version

0.8.16

JamesL6, 2022-09-24:
**### Describe the bug**
Getting "Dashcam Mode - Unsupported Car" on my already supported vehicle. It takes multiple times of turning off the vehicle + rebooting the C3 in order for it to go away. Sometimes, it doesn't happen.

**### Which car does this affect?**
2022 Camry Hybrid XSE

**### Provide a route where the issue occurs**

1. 76fcc86c5aa85c45|2022-09-23--19-21-57--0
2. 76fcc86c5aa85c45|2022-09-23--19-23-32--0
3. 76fcc86c5aa85c45|2022-09-23--19-23-48--0
4. 76fcc86c5aa85c45|2022-09-23--19-25-24--0
5. 76fcc86c5aa85c45|2022-09-23--19-25-45--0

**### openpilot version**
0.8.17

**### Additional info**
The issue first started happening after updating to a recent 0.8.17 update within the last week. I have not experienced this with any other update on my vehicle and C3.

sshane, 2022-09-28:
@jlhsr1 can you provide your dongle id or a route name?

sshane, 2022-09-28:
@JamesL6 Make sure you upload the full rlogs for those routes, it looks like they're gone already unfortunately. I haven't seen a Toyota with this issue before, strange. Please let me know if you have any routes with rlogs uploaded!

jlhsr1, 2022-09-28:
My Dongle ID is 9936b4bbf75a5fe4. The car was unrecognized so I guess that
I would not have a route. I did make trips in my Subaru Ascent which is
recognized but I'm guessing that route information would not be helpful.

sshane, 2022-09-28:
Ah, it looks like your new car is just not in our data base yet! To be clear, you have a 2020 Subaru Ascent that fingerprints properly, but your new car (Subaru Legacy 2021) does not fingerprint?

I'll post a branch here shortly for you to test

sshane, 2022-09-28:
@jlhsr1 Ah, it looks like your new car is just not in our data base yet! To be clear, you have a 2020 Subaru Ascent that fingerprints properly, but your new car (Subaru Legacy 2021) does not fingerprint?

Can you try this branch? https://github.com/commaai/openpilot/pull/25914

Installer URL: https://installer.comma.ai/commaai/missing-legacy-fws

sshane, 2022-09-28:
@JamesL6 You have a separate issue, can you make an issue for that? This issue will be closed soon

jlhsr1, 2022-09-28:
> make an issue for that? This one will be closed soon

Thats correct

jlhsr1, 2022-09-28:
sorry I didn't read it all. I'll try that branch and let you know if I need
more help or it takes care of my problem. Thanks John

jlhsr1, 2022-09-28:
Ok Shane it's working, I installed
https://installer.comma.ai/commaai/missing-legacy-fws but didn't find
Branch # 25914 but am now on openpilot v0.8.17 was on 16. I'm on target
branch missing-legacy-fws. Let me know if all's good.  Thanks again John.

sshane, 2022-09-28:
Merged your fingerprints. Until a new release, you'll want to use the `master` branch

closed 2022-09-28 by sshane

jlhsr1, 2022-09-28:
> master-ci branch

Will do, thanks!

jyoung8607, 2025-09-02:
@GerGa00 We'll need a route from your car to look up what's happening. You can get it from comma connect, just pick any route from your car and go to More Info, there'll be a copy button. The route ID will look like `f39cf149898833ff/00000046--3298c279ff/0`.

GerGa00, 2025-09-05:
> [@GerGa00](https://github.com/GerGa00) We'll need a route from your car to look up what's happening. You can get it from comma connect, just pick any route from your car and go to More Info, there'll be a copy button. The route ID will look like `f39cf149898833ff/00000046--3298c279ff/0`.

I have a fork installed right now(sunnypilot), do I need to install openpilot again to get the route for you?

Thanks

GerGa00, 2025-09-06:
> sunnypilot routes will work for this purpose.

Here is one from the column "route_name"
f5dddb74aa41bd0f/00000010--f3f54d1fa7