commaai/openpilot #25887: Tethering is not persistent

DerGue69, 2022-09-24:
I enabled tethering on my C3 with a self owned SIM card. Which is a cool feature.
At the next drive tethering was disabled and I had to manually enable the feature again.

Would be nice, if tethering is always active when it has been activated. Or at least when the car's engine is on.

morrislee, 2022-09-25:
this is normal operation, if tethering is enabled, C3 will not connect to a wifi hotspot automatically.

I do however, wish it would auto enable hotspot when its on road, and disable it when parked

adeebshihadeh, 2023-01-02:
This is how it's intended to operate right now. In the future, we can add some more options here, but this is a bit tricky without complicating the UI since I've seen lots of different (valid) ways people want this to work.

closed 2023-01-02 by adeebshihadeh