commaai/openpilot #26016: GM camera ACC: LKAS fault when relay closes/opens

sshane, 2022-10-08:
- [ ] Fault when relay opens (we send counter of 1, should copy camera's counter) - https://github.com/commaai/openpilot/pull/26013
- [ ] Fault when relay closes (our counter doesn't sync up with camera's)

ncoop23, 2022-10-26:
0def4a390f6fe5c0|2022-10-25--19-17-40--0

Just had this happen today driving along straight and level near the end of the drive. Also had a little steering tune issue as well.

dominicdoty, 2022-11-05:
I'm in a 2017 Bolt on an opgm/newcombo variant, but in case it helps your debugging, I had an LKAS fault while driving today on the segments:
`82a0516f27e12844\|2022-11-04--18-03-04--15`
`82a0516f27e12844\|2022-11-04--18-03-04--16`

abueide, 2022-12-01:
@sshane we ran into this on our bolt euv and it scared the crap out of us xD. Is there anything we can do to help fix this? We're pretty new to openpilot development so any advice would be appreciated!

minorOffense, 2022-12-08:
Just happened to me today on the bolt-torque branch (0.9.1) on a 2022 Bolt EUV

dominicdoty, 2022-12-09:
Had another LKAS fault recently, my branch which is synced to comma master LKAS sync code.

I've had LKAS faults on this exact segment of road under similar conditions 3 or 4 times, leading me to think there's some reproducible issue there.

Its a long left hand sweeping on ramp with a meter light at the end. OP disengaged, heavy traffic, stop and go at 0-10 mph.

`82a0516f27e12844|2022-12-08--17-07-17--22`

ncoop23, 2022-12-13:
It seems these errors are happening more often on the EUV. They do not clear on a vehicle power cycle sometimes. I turn off the car, open them close the door, wait for the dash to go dark and then power on the car. The comma still says LKAS fault. Then I drive for a while as I have somewhere to go and don't care about the comma working or not and the cruise control won't work so I unplug the comma and there's a terribly annoying beeping and the cruise control works. This is very frustrating and makes me want to return the comma since at this point about 25% of my drives, it doesn't work.

abueide, 2022-12-13:
@ncoop23 you need to wait at least 3 minutes between closing the door and turning the bolt back on, seems to fix the error every time for me at least. This is cause the car doesn't completely shut off until after 3 minutes of being "off".

sshane, 2022-12-13:
That's correct, I've observed 2-3 minutes with the car fully off to reset the fault. @ncoop23 when do these faults mostly happen for you? While driving? When first turning the car on? Or shortly after turning the car off?

sshane, 2022-12-13:
@dominicdoty I've requested upload, can you put your device on wifi?

@abueide let me know if you have a segment where it happens and I can take a look

ncoop23, 2022-12-13:
> That's correct, I've observed 2-3 minutes with the car fully off to reset the fault. @ncoop23 when do these faults mostly happen for you? While driving? When first turning the car on? Or shortly after turning the car off?

I generally get them when driving for a while.  Some have been near turns, other on the straight and level.  I think there may be a buffer overflow or a command that needs an ack that doesn't get it like a calibration or check every so often.  About to find and upload the latest one. Car is syncing. They are random. The strange thing is something latches as if I disconnect the comma there is a terribly annoying beeping coming from somewhere and I don't know if it's the comma.

dominicdoty, 2022-12-14:
@sshane Looks like the first one I posted I must have unpreserved, the one from 12/8 should have uploaded now - apologies for the delay.

Let me know if there's anything I can do to help out. These errors are pretty rare for me, but would be nice to get them fixed.

jelimoore, 2022-12-15:
@sshane this is @abueide 's recent LKAS failure route ID: `c9dc069b00b41849|2022-12-14--14-11-59--26`

dominicdoty, 2022-12-16:
another one, back to opgm fork, exact same location and conditions
`82a0516f27e12844|2022-12-15--18-01-41--13`

minorOffense, 2022-12-23:
Happened to me again at 10:22 to 10:23am on Thursday dec 22nd. 

6afbe9126b0b488e

I went to look for the video on connect but it’s not showing up. Maybe the logs still exist but I can see them. If it shows up I’ll post a proper link here.

minorOffense, 2023-01-02:
Got one uploaded finally. 

6afbe9126b0b488e|2023-01-02--12-35-07--2

Check around 12:37 that’s when it triggered. 

I set the flag on connect to preserve this drive.

minorOffense, 2023-01-03:
And another at 10:56

6afbe9126b0b488e|2023-01-03--09-42-55--0

minorOffense, 2023-01-03:
And again at 13:58

6afbe9126b0b488e|2023-01-03--12-54-29--0

ncoop23, 2023-01-03:
So what happens to correct this issue?  It's kind of a big deal since it causes the comma to abort and need a car restart.

sshane, 2023-01-17:
I've been busy with other work recently, but will look at all the issues. Make sure if you report a segment to upload all logs and preserve

sshane, 2023-01-18:
@minorOffense I can't guarantee this will work, it's just a hypothesis I have, but can you try this branch?

https://github.com/commaai/openpilot/tree/gm-full-rate-fault
Installer URL: https://installer.comma.ai/commaai/gm-full-rate-fault

The two faults you posted the car had a temporary fault followed by a permanent fault, it could be possible that openpilot slowing its send of the inactive steer frame could be exacerbating the issue during the temp fault, leading to a perm fault.

Please drive on it for a few days and let me know with a route if you get any more temporary faults, and if it recovers.

ncoop23, 2023-01-20:
> @minorOffense I can't guarantee this will work, it's just a hypothesis I have, but can you try this branch?
> 
> https://github.com/commaai/openpilot/tree/gm-full-rate-fault Installer URL: https://installer.comma.ai/commaai/gm-full-rate-fault
> 
> The two faults you posted the car had a temporary fault followed by a permanent fault, it could be possible that openpilot slowing its send of the inactive steer frame could be exacerbating the issue during the temp fault, leading to a perm fault.
> 
> Please drive on it for a few days and let me know with a route if you get any more temporary faults, and if it recovers or not.

I'm driving around on this now. So far so good. The steering tuning isn't damped enough but that's something else.

ncoop23, 2023-01-21:
Had what liked like 2 steering errors in quick succession on your test build yesterday. Nothing caused comma to fault, only warn. 0def4a390f6fe5c0|2023-01-20--21-51-56--12

minorOffense, 2023-01-23:
I’ve installed it and I’ll let you know.

jettoblack, 2023-01-24:
@sshane I have been testing this change (applied to another branch of SunnyPilot) for a few days and thought the LKAS fault had been resolved, unfortunately just now I got the error TAKE CONTROL NOW! LKAS Fault - Restart car to engage.

48c21cb1105d4d1c|2023-01-23--19-30-58--5

Happened around 19:36:10. I was driving mostly straight at the time. 2022 Bolt EUV. Let me know if there's anything else I can do to help.

ncoop23, 2023-01-24:
Got a take control now straight and level on the freeway with this branch. 

0def4a390f6fe5c0|2023-01-22--20-13-34--70

minorOffense, 2023-01-27:
I’ve been stuck at home with COVID all week so I haven’t had a chance to test. Hopefully next week I’ll get a chance.

sshane, 2023-01-29:
I have a branch now to reduce the send rate of the LKAS message to the EPS, as we suspect that sending two LKAS messages too close together is causing faults (undetectable by openpilot at just 100Hz). This branch changes the 50Hz message down to just 33Hz, so steering may be impacted, but if this hypothesis is correct, faulting should be reduced.

Installer URL: https://installer.comma.ai/commaai/gm-reduce-rate

https://github.com/commaai/openpilot/tree/gm-reduce-rate

fredcredeur, 2023-01-29:
Are you able to see how many people are currently using a branch? Just curious. 

I am installing the gm reduce rate branch and will report issues.

ncoop23, 2023-01-29:
> I have a branch now to reduce the send rate of the LKAS message to the EPS, as we suspect that sending two LKAS messages too close together is causing faults (undetectable by openpilot at just 100Hz). This branch changes the 50Hz message down to just 33Hz, so steering may be impacted, but if this hypothesis is correct, faulting should be reduced.
> 
> Let me know how this branch does @ncoop23 @jettoblack @minorOffense
> 
> Installer URL: https://installer.comma.ai/commaai/gm-reduce-rate
> 
> https://github.com/commaai/openpilot/tree/gm-reduce-rate

I can load it to try. Did you see that in mine that double errored?

sshane, 2023-02-01:
I just merged two fixes to help significantly reduce LKAS faults when starting and stopping your vehicle a few times. Please try out `master` branch and let me know if those are gone now! https://github.com/commaai/openpilot/pull/27164

fredcredeur, 2023-02-01:
> I just merged two fixes to help significantly reduce LKAS faults when starting and stopping your vehicle a few times. Please try out `master` branch and let me know if those faults are gone now! #27164

Will you update the GM Reduce Branch or should we uninstall this branch to try out the master branch? On my 3rd day with GM Reduce Branch and have had zero LKAS Faults.

sshane, 2023-02-01:
Haven't tried that branch yet myself, how is the steering control? Hopefully it should be the same

fredcredeur, 2023-02-02:
All week so far I thought the steering felt better.  Actually noticed and thought about it multiple times that it just felt perfect like it centered the lane very well. Until my drive home a while ago. For some of the trip it felt like it just kept hugging the line opposite of the center line. A couple of times I made a point to nudge the vehicle back center and it would move back. Made a point to try this in the left lane and right lane and it did the same thing in both. While it did this it felt like it needed to make more corrections (like it ping-ponged a bit more).  Maybe it was the weather. Roads are wet today? All week up until my drive home just a while ago id say it's been flawless. Even with the hugging of the line Im satisfied without getting that LKAS Fault.

Verylukyguy, 2023-02-02:
I know that I am not a Camera Car, but I just loaded a fresh fork of Comma Master and I am getting a LKAS Fault: Restart Car to Engage error on startup that clears on its own without a restart.

sshane, 2023-02-02:
Can you send a segment @Verylukyguy?

sshane, 2023-02-02:
@fredcredeur updated that branch. Please try that + the lower steering message rate for the next week to stress test it

sshane, 2023-02-02:
@fredcredeur can you send your dongle ID?

minorOffense, 2023-02-03:
I got three temporary failures in a row but they didn’t trigger the lkas fault which is good. But yeah each time I tried to rengage the same cruise setting it would temporary fault and I’d take over. 

I’ll upload the logs and maybe there’s something about the cruise state I was in that it didn’t like (speed maybe had weird rounding or decimal errors, who knows). Cause once I changed the speed it stopped temporary faulting.

sshane, 2023-02-03:
> I got three temporary failures in a row but they didn’t trigger the lkas fault which is good. But yeah each time I tried to rengage the same cruise setting it would temporary fault and I’d take [...]

Please send a route with issues so I can check it out

minorOffense, 2023-02-03:
6afbe9126b0b488e|2023-02-02--19-17-59--0

There was three faults on this route. Around 19:20 I think they started. I was actually on the old gm branch you had created though at the time.

sshane, 2023-02-09:
Hey everyone, the random LKAS faults should be fixed in `master`, `master-ci`, and `nightly` branches. Please let us know if you try these branches and see anything strange!

closed 2023-02-10 by sshane

minorOffense, 2023-02-10:
I’m still on that other branch and got some process communication failures. Not sure if it’s related. 

6afbe9126b0b488e|2023-02-09--10-04-26--0

Anyways I’ll switch back to master and test there too. Thanks.

ncoop23, 2023-03-10:
Got another straight line red error a couple days ago:
0def4a390f6fe5c0|2023-03-09--10-36-17--93
I posted in tuning, is there a good place to be able to get better at describing an error?  The car isn't always in a place where I can SSH into it but I am connected to connect to pull logs and such.

sshane, 2023-03-10:
@ncoop23 that's not an LKAS fault, that's a temp ACC fault. With an LKAS fault you would not have been able to re-engage. Looks like this ACC fault was caused by panda keeping controls allowed high when you pressed the main button to disengage, then re-engaged 2 seconds later. Cross posted to this issue to track this: https://github.com/commaai/openpilot/issues/27479

kjgarrett23, 2024-06-09:
Hey, I’ve got a 2020 Silverado that uses sunny pilot. I get LKAS when driving long ways for work. It’ll pop up anytime 45 min to two hours but will always happen on a drive longer than 2 hours. Any thoughts?