commaai/openpilot #26038: Investigate possible source of brake pressure in Hyundais

andrew-ledawson, 2022-10-12:
[As noted in the Hyundai car state parser](https://github.com/commaai/openpilot/blob/741867813285672a723b8fc53ead65a5cbe5c6dd/selfdrive/car/hyundai/carstate.py#L99), brake pressure detection is not supported at this moment.  But [CYL_PRES from ESP12](https://github.com/commaai/opendbc/blob/a70a90016ca0a4af844116ff6f2723f9687ecfff/hyundai_kia_generic.dbc#L1227) might have the right data, considering it is part of the stability control system and is measuring some pressure in bars.  [Hyundai diagnostic PDFs](https://static.nhtsa.gov/odi/rcl/2013/RCRIT-13V489-0349.pdf) show that their diagnostic software also measures brake pressure in bar.  And "CYL_PRES" sounds like is could be short for "brake cylinder pressure".  A developer with access to a Hyundai vehicle could test this and see if that value corresponds to brake pressure.

gregjhogan, 2022-10-12:
From what I have seen this signal is brake pressure, but only when the brake pedal is pressed (always zero for electronic actuation).

adeebshihadeh, 2022-10-12:
Thanks for the interest, but we don't create issues for TODOs like this so as not to pollute our issue tracker. The TODOs are sufficient for cases like this.

closed 2022-10-12 by adeebshihadeh