commaai/openpilot #26046: 2023 Hyundai Elantra Limited Hybrid Note Supported Vehicle

ashishlalani, 2022-10-12:
### Describe the bug

It says that Dashcam mode is active and says unrecognized car when I try to use it.
It is a 2023 Hyundai Elantra Limited Hybrid (and supports SCC).
I can provide what is needed to have it supported as 2022 model is supported. 

### Which car does this affect?

2023 Hyundai Elantra Hybrid Limited

### Provide a route where the issue occurs

All Routes

### openpilot version

Newest Version

incognitojam, 2022-10-12:
You need to provide your dongle ID and a route name so that the car firmware can be collected

ashishlalani, 2022-10-12:
Dongle ID: [8dcf421697cd2cb0](https://useradmin.comma.ai/?onebox=8dcf421697cd2cb0)
Route Name: [8dcf421697cd2cb0|2022-10-12--16-12-21](https://useradmin.comma.ai/?onebox=8dcf421697cd2cb0%7C2022-10-12--16-12-21)

ashishlalani, 2022-10-12:
Please let me know when you can get it figured out to add. If you can give me the updated values.py file, I am really having trouble with Step 4 in the fingerprinting guide and can't really figure it out, I can attempt to SSH into the comma and overwrite the existing one, and hopefully you can officially add support on it soon as well!

Thanks

incognitojam, 2022-10-12:
Can you try factory resetting and installing `installer.comma.ai/commaai/fw-hyundai-elantra-hev-2023`?

sshane, 2022-10-13:
@ashishlalani can you make sure your device stays online for a short while so I can upload a log?

ashishlalani, 2022-10-13:
It looks like it works with that branch but longitudinal is done by smart cruise control and when I activate the e2e it keeps slowing down and braking hard. Thanks so much. It should be online for the next 12 hours or so so you can use the logs of the most recent drive

ashishlalani, 2022-10-13:
Route Name: 8dcf421697cd2cb0|2022-10-12--21-15-47

RaunakThomas, 2022-10-13:
@ashishlalani I have a Elantra 2023 Hybrid limited edition and I actually have pushed the values.py file to my car using SSH, its been working for the past couple of weeks but I didn't really know how to do the pull request here, so thanks for that! Hope you folks add it to future releases because I had to do it today again after an update as I'm on the master branch. PFB the text for the values.py that I'm using. You can contact me in discord if you are facing issues. (RaunakThomas #4223)

 CAR.ELANTRA_HEV_2021: {
    (Ecu.fwdCamera, 0x7c4, None) : [
      b'\xf1\x00CN7HMFC  AT USA LHD 1.00 1.07 99210-AA000 220426',
      b'\xf1\x00CN7HMFC  AT USA LHD 1.00 1.05 99210-AA000 210930',
      b'\xf1\000CN7HMFC  AT USA LHD 1.00 1.03 99210-AA000 200819',
    ],
    (Ecu.fwdRadar, 0x7d0, None) : [
      b'\xf1\000CNhe SCC FHCUP      1.00 1.01 99110-BY000         ',
      b'\xf1\x8799110BY000\xf1\x00CNhe SCC FHCUP      1.00 1.01 99110-BY000         ',
    ],
    (Ecu.eps, 0x7d4, None) :[
      b'\xf1\x00CN7 MDPS C 1.00 1.03 56310BY0500 4CNHC103',
      b'\xf1\x8756310BY0500\xf1\x00CN7 MDPS C 1.00 1.03 56310BY0500 4CNHC103',
      b'\xf1\x8756310/BY050\xf1\x00CN7 MDPS C 1.00 1.03 56310/BY050 4CNHC103',
      b'\xf1\x8756310/BY050\xf1\000CN7 MDPS C 1.00 1.02 56310/BY050 4CNHC102',
    ],
    (Ecu.transmission, 0x7e1, None) :[
      b'\xf1\0006U3L0_C2\000\0006U3K3051\000\000HCN0G16NS0\xb9?A\xaa',
      b'\xf1\0006U3L0_C2\000\0006U3K3051\000\000HCN0G16NS0\000\000\000\000',
      b'\xf1\x006U3L0_C2\x00\x006U3K3051\x00\x00HCN0G16NS0\xb9?A\xaa',
      b'\xf1\x816U3K3051\000\000\xf1\0006U3L0_C2\000\0006U3K3051\000\000HCN0G16NS0\xb9?A\xaa',
      b'\xf1\x816U3K3051\x00\x00\xf1\x006U3L0_C2\x00\x006U3K3051\x00\x00HCN0G16NS0\x00\x00\x00\x00',
      b'\xf1\x816U3K3051\x00\x00\xf1\x006U3L0_C2\x00\x006U3K3051\x00\x00HCN0G16NS0\xb9?A\xaa',
    ],
    (Ecu.engine, 0x7e0, None) : [
      b'\xf1\x816H6G5051\x00\x00\x00\x00\x00\x00\x00\x00',
      b'\xf1\x816H6G6051\x00\x00\x00\x00\x00\x00\x00\x00',
    ]
  }

ashishlalani, 2022-10-13:
One more thing I noticed is that the speed limit recognition sign assist stops working and doesn't display the current speed limit anymore when Comma is plugged in as well.

closed 2022-10-13 by sshane

braidbunch, 2023-03-25:
I have the same problem. I did the factory reset and tried to install the url you provided, installer.comma.ai/commaai/fw-hyundai-elantra-hev-2023. It acted like it was going to download and then went right back to the screen asking for a url again. I tried again with same results. It’s a 23 Hyundai Elantra Hybrid Limited. I got the unit yesterday. Still get car unrecognized with the original url. Please help!

RaunakThomas, 2023-03-25:
I don’t remember the actual link but the latest version on GitHub actually
supports the Elantra hybrid. So you should be able to use the latest
standard link and it should work

PR #26055 (2022-10-13): Hyundai: add missing Elantra Hybrid 2023 FW versions
`8dcf421697cd2cb0|2022-10-12--16-12-21--0`

- [x] waiting for tester

closes #26046