commaai/openpilot #26067: Bolt EUV turns in late

ncoop23, 2022-10-13:
### Describe the bug

Car turns into a turn late and then over corrects which requires extra torque and then sometimes it gets jerky/underdamped. 

### Which car does this affect?

2022 Bolt EUV Premier non Super Cruise

### Provide a route where the issue occurs

0def4a390f6fe5c0|2022-10-08--12-57-26--4

### openpilot version

0.8.16

### Additional info

Around timestamp 800 car is going about 40 and comes up on a turn which had good visibility prior.  It reacts late and has to cut hard to make the turn and goes out of the lane.

podony, 2022-10-17:
Which brunch should I use on euv/comma2 ?

sshane, 2022-10-17:
We're looking into fixing the Bolt's steering relatively soon, I'll let you know when we have something merged in master

ncoop23, 2022-10-25:
Another instance flagged.

This was on 8.17 master
0def4a390f6fe5c0|2022-10-25--16-31-35--21

ncoop23, 2022-10-25:
Almost put me into a curb near the end of this one. 

0def4a390f6fe5c0|2022-10-25--17-44-38--0

ncoop23, 2022-11-09:
It almost went off the road yesterday. Latest master-ci build

0def4a390f6fe5c0|2022-11-08--04-42-54--8

ncoop23, 2022-11-09:
https://user-images.githubusercontent.com/1179525/200882184-25b6e417-6d16-417a-b7d7-ff6a6a9a7f22.mp4

Video of the latest issue

sshane, 2022-11-30:
Duplicate of https://github.com/commaai/openpilot/issues/25352, feel free to post there. We have a new branch you can try here: https://github.com/commaai/openpilot/pull/26371. Please post any feedback you have!

https://installer.comma.ai/commaai/bolt-torque-control

minorOffense, 2022-12-02:
I just got my Comma 3 and I have a Bolt EUV. I'm testing this out and will report back.

ncoop23, 2022-12-02:
> We have a new branch you can try here: #26371. Please post any feedback you have!
> 
> https://installer.comma.ai/commaai/bolt-torque-control

Trying now

closed 2022-12-02 by ncoop23

minorOffense, 2022-12-04:
Yeah it definitely cuts the inside of a turn. On ramps and low speed sharp turns especially.

minorOffense, 2022-12-07:
Also something to note I noticed the speedometer on the car is usually 2KM/hr higher than what the comma is showing. Not sure if that factors into the equation when dealing with turning.