commaai/openpilot #26094: 2018 Kia Stinger GT2 - Car Unrecognized Often since 0.8.16

aegisx5, 2022-10-16:
### Describe the bug

Since the latest update to Openpilot 0.8.16 I have been frequently getting error: car unrecognized and goes into dashcam mode. Sometimes it's fixed by restarting the C3 and the car, often not. Happens about 60% of the time. Tried reinstalling the main Openpilot branch without success.

### Which car does this affect?

2018 Kia Stinger GT2

### Provide a route where the issue occurs

1ce52d9487767eae|2022-10-15--15-03-44

### openpilot version

0.8.16

sshane, 2022-10-17:
It looks like you don't have your comma power plugged in, you're purely CAN fingerprinting. Can you try to plug in the comma power?

aegisx5, 2022-10-17:
I'm not entirely sure what you mean. It goes USB-C for power to the piece that then divides off into Ethernet (going to the OBDII) and to the LKA cameras. Wouldn't it not turn on if there was no power?

Should I try a new USB-C cable?

sshane, 2022-10-17:
Yes, if you have a comma power already installed, try to re-seat all connections and make sure you're not using an OBD splitter.

aegisx5, 2022-10-17:
I am using an OBDII splitting, but it’s been working fine that way with my C2 and now my C3. I have a piggyback tuner that also needs OBDII access all the time. Any suggestions on how to connect both?

sshane, 2022-10-17:
Oh I checked your routes out and it looks like you were always CAN fingerprinting. So something recently has made it more inconsistent, possibly our raising of the CAN fingerprint time gathering more addresses. I would try without the splitter to make sure that's the issue.

aegisx5, 2022-10-27:
I changed out the Ethernet cable for CAT8 and a new higher quality splitter (I need the splitter for my tuner). So far the problem seems to be resolved, although I only tested it on a few drives and the issue was sporadic so I am still verifying. I wonder if the flat Ethernet cable failed - might be good to just use CAT8 with all of them. Better interference resistance and less signal loss.

sshane, 2022-10-28:
It looks like since the 22nd you haven't had any issues. Shall we close this issue?

closed 2022-11-11 by adeebshihadeh