commaai/openpilot #26139: UI: segfault when switching camera streams

HaraldSchafer, 2022-10-18:
### Describe the bug

When switching camera streams in the ui, as is done in this commit.
https://github.com/commaai/openpilot/commit/1db0b6acf0ca433a2eeaec4ed1af6d51241e9061

It can sometimes segault.

It's relatively easy to reproduce by checking out that commit and running replay on public route
```./replay "1df037e2a95eb9be|2022-10-17--14-52-31" --ecam ```

If you then skip forward and back frequently you should quickly see a segault

```batman@workstation-research-harald:~/openpilot$ ./selfdrive/ui/ui 
Segmentation fault (core dumped)
```

### Provide a route where the issue occurs

1df037e2a95eb9be|2022-10-17--14-52-31

### openpilot version

draw-mpc

HaraldSchafer, 2022-10-18:
@deanlee Something you want to look into?

deanlee, 2022-10-24:
The segfault I encountered was :

> #0  0x00007f6676881cd0 n/a (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so + 0x109ecd0)

It's a opengl problem, iris_dri is part of libgl1-mesa-dri (the OpenGL API -- DRI modules):
https://packages.ubuntu.com/search?suite=focal&arch=amd64&mode=exactfilename&searchon=contents&keywords=iris_dri.so

closed 2022-10-27 by adeebshihadeh