commaai/openpilot #26176: HKG: steering temporarily unavailable (90 degree steering unlock didn't work?)

tecandrew, 2022-10-20:
### Describe the bug

Was testing e2e behavior and made a left turn at intersection (stop light). On green light it began to go slowly, so I pressed accel manually, and OP began turning. The wheel suddenly jumped back to normal position (to 0 degrees/straight), "Steering Temporarily Unavailable" showed up on the UI, and I took control. 

https://user-images.githubusercontent.com/21319730/197001150-9e52cf6e-31b8-41c5-8d8d-0b3dcc273590.mp4

### Which car does this affect?

Hyundai Elantra Hybrid 2021

### Provide a route where the issue occurs

eb5a50f9e98f50b9|2022-10-20--08-12-57--26

### openpilot version

0.8.17-master-ci

sshane, 2022-10-21:
I've never seen a fault from our Sonata or Optima, applying torque for very long periods of time with this fix. Can you try to reproduce this again to make sure the EPS is not respecting our workaround for the 90 degree fault?

For testing only, you can modify latActive to ignore active, and allow all messages in panda safety. Then you can sit in park with the wheel turned past 90 for a while.

[image]

tecandrew, 2022-10-26:
haven't found the time to test constant steering using OP, but it occurred again. funny enough it happened at the exact intersection. `eb5a50f9e98f50b9|2022-10-25--17-32-43--13`

https://user-images.githubusercontent.com/21319730/198123080-3dc81253-46ad-4085-b73b-92f07b20858c.mp4

sshane, 2022-10-28:
Can you try this PR and let me know how it goes? https://github.com/commaai/openpilot/pull/26275

How often has this happened so far? Just twice in a week so far?

tecandrew, 2022-10-28:
it's happened every time on that intersection. I can't reproduce it anywhere else

will test today

tecandrew, 2022-10-28:
separate issue, but the branch change menu/feature on the UI put my C3 out of commission for now

adeebshihadeh, 2022-10-29:
> separate issue, but the branch change menu/feature on the UI put my C3 out of commission for now

What do you mean?

tecandrew, 2022-10-29:
it's fixed now, but hereMs what I did while on master-ci: 

1. switched branches on the UI menu
2. selected the branch from the show list
3. pressed download
4. pressed install
5. rebooted and it got stuck on comma logo

I SSH'd to see what's up and none of the submodules were initialized. couldn't check out another branch because the submodules were configured weird. I just a fresh pull of OP to fix it 😅

sshane, 2022-10-31:
@tecandrew so far you said this has happened every time at that same intersection? How many times is that? Has the 3-frame fix worked?

adeebshihadeh, 2022-10-31:
> here's what I did while on master-ci:
> 
> 1. switched branches on the UI menu
> 2. selected the branch from the shown list
> 3. pressed download
> 4. pressed install
> 5. rebooted and it got stuck on comma logo when it powered on
> 
> I SSH'd to see what's up and none of the submodules were initialized. couldn't check out another branch because the submodules were configured weird. I just did a fresh pull of OP to fix it sweat_smile

Have you done anything weird with your git config or anything like that? Were you switching between branches with and without submodules?

tecandrew, 2022-10-31:
@sshane I actually _just_ tried it and it faulted in a similar manner on a different area (right turn this time). `eb5a50f9e98f50b9|2022-10-31--12-44-32--1`

> How many times is that?

It's only happened twice that I've mentioned it here.

https://user-images.githubusercontent.com/21319730/199114201-4236375d-5dc1-486e-b771-233fd58c0df3.mp4

tecandrew, 2022-10-31:
@adeebshihadeh I haven't done much other than keep a separate directory for `master-ci`. After the switch-branch hiccup I SSH'd to see what's up then tried to switch back to `master-ci` manually and couldn't.

I usually just SSH to see if it's on my network; development is done on my linux machine. 

```
comma@tici:/data$ ls -l
total 200
-rwxr-xr-x   1 comma comma     63 Aug 31 09:22 continue.sh
drwxrwx---   3 root  root    4096 Apr 21  2022 data
drwxr-xr-x   4 root  root    4096 Oct 31 14:31 etc
drwxr-xr-x   2 comma comma 126976 Oct 31 14:24 log
drwxr-xr-x   4 comma comma   4096 Aug 31 09:26 media
drwxr-xr-x   3 root  root    4096 Apr 21  2022 misc
drwxr-xr-x  21 comma comma   4096 Oct 31 13:34 openpilot
drwxr-xr-x  20 comma comma   4096 Oct 30 11:26 openpilot.master-ci
drwxr-xr-x   3 comma comma   4096 Oct 31 14:32 params
drwx------   4 root  root    4096 Apr 21  2022 persist
lrwxrwxrwx   1 comma comma     15 Oct 31 13:33 pythonpath -> /data/openpilot
drwxr-xr-x   5 comma comma   4096 Oct 31 13:34 safe_staging
drwxr-xr-x 259 comma comma   4096 Oct 25 18:08 scons_cache
drwxr-xr-x   2 comma comma   4096 Apr 21  2022 ssh
drwxr-xr-x   2 comma comma  20480 Oct 31 14:31 stats
drwxr-xr-x   3 comma comma   4096 Apr 21  2022 tmp
```

tecandrew, 2022-11-01:
@adeebshihadeh I encountered the branch-change bug again. Filed a separate issue here #26314

sshane, 2022-11-22:
@tecandrew have you experienced the steering workaround code failing at all recently?

tecandrew, 2022-11-23:
haven't experienced the `steering unavailable` lately on master-ci 🤔 I just went on the same route/intersection as the one in the initial report and experienced only the steering/torque max warning

sunnyhaibin, 2023-01-13:
User reported that the EPS still faulted with his 2019 Hyundai Ioniq when his car tried to steer over 90 degrees. Even though the EPS faulted multiple times during that turn, it was still able to continue to command torque past 90 degrees.
- `012c95f06918eca4|2023-01-13--22-00-59--3`

sunnyhaibin, 2023-01-13:
@sshane User reported that the EPS still faulted with his 2019 Hyundai Ioniq when his car tried to steer over 90 degrees. Even though the EPS faulted multiple times during that turn, it was still able to continue to command torque past 90 degrees.
- `012c95f06918eca4|2023-01-13--22-00-59--3`

andy741217, 2023-01-14:
@sshane  The EPS still faulted when steer over 90 degrees then openpilot can control over 90 degrees

- 012c95f06918eca4|2023-01-14--12-08-04--6

sunnyhaibin, 2023-01-15:
`012c95f06918eca4|2023-01-14--12-08-04`

The current frequency works for this car to a certain point it seems:
[image]

andy741217, 2023-01-20:
@sshane older hkg can't unlock steer fault but i try it many times

- 012c95f06918eca4|2023-01-20--15-49-52--4
[https://streamable.com/eu3aks](url)

sshane, 2023-01-20:
@andy741217 do you want to test this branch? Your car's EPS faulted earlier than our fault workaround expected. Please avoid touching the wheel as there is a different bug for overriding when above 90 deg that you may now hit easier with these limits.

https://github.com/commaai/openpilot/tree/h90d-legacy
Installer URL: https://installer.comma.ai/commaai/h90d-legacy

andy741217, 2023-01-21:
When I install h90d-legacy then reboot it.I got "openpilot failed to build"

andy741217, 2023-01-21:
@sshane h90d-legacy always show openpilot fail to build,so i install master-ci and motify it,

- 012c95f06918eca4|2023-01-21--10-39-50--3
steer can 90degree  but will show warning

tecandrew, 2023-02-09:
bug occurred again on the latest version of `master-ci`:

```
eb5a50f9e98f50b9|2023-02-07--18-49-24--12
```

i don't have the render tools on hand atm to create the video with overlay.

tecandrew, 2023-02-14:
https://user-images.githubusercontent.com/21319730/218649274-bae019dc-e04e-4a78-b5ca-2ddd60c03519.mov

Recent fix to connect.comma.ai allowed me to get a decent screen grab of the behavior.

tecandrew, 2023-02-14:
another occurence: `eb5a50f9e98f50b9|2022-12-22--18-16-18--28`

https://user-images.githubusercontent.com/21319730/218653943-e1e8303b-5b38-458c-b26e-f7e726942149.mov