commaai/openpilot #26179: kia seltos 2021

Alejoaguirre3a, 2022-10-20:
### Describe the bug

I just received my open pilot comma ai 3 but doesn't recognized my car, my car was in the list so why i do have issues?
KIA SELTOS MODEL S 2021. commaai works only as a dash cam mode

### Which car does this affect?

KIA SELTOS S 2021

### Provide a route where the issue occurs

na

### openpilot version

updated

### Additional info

N/A

sshane, 2022-10-21:
Can you provide your dongle ID? It's in device settings

Alejoaguirre3a, 2022-10-21:
> Can you provide your dongle ID? It's in device settings

724f06bcdb81bc8e

sshane, 2022-10-21:
I only see Kia Stinger routes in that dongle id, are you sure it's yours?

Alejoaguirre3a, 2022-10-21:
Yes it is I bought it to used in two cars. the kia stinger doesn't have a problem.but when i connected to the kia seltos  2021 does not work.

Alejoaguirre3a, 2022-10-21:
[image]
[image]
[image]

this are the pictures to show you the messages from the comma ai screen .

sshane, 2022-10-21:
It's not showing up in useradmin for some reason. Make sure to plug it into the Seltos, and turn the car on for a few minutes to ensure the C3 saves the route

Alejoaguirre3a, 2022-10-21:
Ok will do!

sshane, 2022-10-21:
I see some routes from last month now, it looks like your radar isn't responding (even to tester present). Do you happen to know if your car lacks a radar?

sshane, 2022-10-21:
From your VIN, it looks like you have a series S. In that case, only the SX has Smart Cruise Control w/ Stop and Go: https://cdn.dealereprocess.org/cdn/brochures/kia/2021-seltos.pdf

You can try a fork to add LKAS to this car, however we cannot support it upstream if it does not have a radar, nor adaptive cruise control.

sshane, 2022-10-21:
Added your Stinger's FP: https://github.com/commaai/openpilot/pull/26198

closed 2022-10-21 by sshane

Alejoaguirre3a, 2022-10-21:
> From your VIN, it looks like you have a series S. In that case, only the SX has Smart Cruise Control w/ Stop and Go: https://cdn.dealereprocess.org/cdn/brochures/kia/2021-seltos.pdf
> 
> You can try a fork to add LKAS to this car, however we cannot support it upstream if it does not have adaptive cruise control. Let me know if you do actually have SCC and I can take a look at this again

I just connected.  My car have cruise control and line assistance as well.  If I get to a stop I have to stop the car my self and then hit the gas to keep moving.  So I don't have stop and go I guess.  [image]

Thanks

sshane, 2022-10-21:
Requested upload, can you put the device online? Does your car actively brake for cars, or is it just normal cruise control? I don't see a distance setting.

Alejoaguirre3a, 2022-10-21:
No it doesn't break for cars, only gives me a sound alert.

sshane, 2022-10-22:
Yep, that means you don't have Smart Cruise Control which is a pre-requisite. Again, forks may offer what you're looking for.