commaai/openpilot #26214: Comma 3 random restart during drive

JamesL6, 2022-10-23:
### Describe the bug

Toward the end of my drive, the Comma 3 randomly restarted, errors were thrown on my dash, then the Comma 3 turned back on, and the dash errors went away. This was all happening while driving.

Car: 2022 Camry Hybrid XSE

Version: Master 0.8.17

Device: Comma 3

### Provide a route where the issue occurs

76fcc86c5aa85c45|2022-10-21--22-19-07--10

### openpilot version

Master - 0.8.17

adeebshihadeh, 2022-10-23:
Thanks for the report. Unfortunately, this is a case of #24302. It's a rare crash, but if it happens again, you can try re-seating the NVMe.

closed 2022-10-23 by adeebshihadeh

JamesL6, 2022-10-24:
Thank you @adeebshihadeh

JamesL6, 2022-11-07:
C3 froze and rebooted again mid drive

76fcc86c5aa85c45|2022-11-04--21-09-42--6