commaai/openpilot #26222: 2023 Hyundai Ioniq 5 HDA 2 OP won't control steering

jdtram, 2022-10-24:
### Describe the bug

I've been working via discord with Sunnyhaibin and a few others on this issue. I added the camera firmware info to my vaules.py and the car is now recognized but steering is not controlled when I engage OP in my 2023 Ioniq 5 with HDA 2. 

I'm using Master-ci branch. I've swapped the USB C cable from the power harness to the Comma 3, the USB A to A cable from the other harness box to the Red panda, the OTG adapter connecting the USB A to the Comma, swapped wires on the Hyundai Q to match Hyundai K, pulled Hyundai K harness from my Elantra and swapped wires to match Hyundai Q. I've tried Sunnypilot branch as well. When engaging OP with OP Long enabled I get a harness relay error.

@adeebshihadeh said he would check this out today. 

@sunnyhaibin created PR #26145 to have the firmware version added for my car

### Which car does this affect?

Hyundai Ioniq 5 2023

### Provide a route where the issue occurs

8a53aa70d88313f1|2022-10-24--09-23-26--0

### openpilot version

0.8.17

briskspirit, 2022-10-26:
` the USB A to A cable from the other harness box to the Red panda` - if I understood you right - you might have burnt transceiver(s).
Open red panda case and take a closeup photo of the top side (where both USB connectors).

jdtram, 2022-10-26:
I never connected the usb A to A to the harness box with 12v power (cat5). I will run out and take pictures though.

jdtram, 2022-10-26:
[image]
[image]

jdtram, 2022-10-26:
I worded that sentence strange because I was typing while trying to remember everything I had done. The first harness (the power harness going to the C3)  got the USB C cable swapped and I swapped the other harness box's USB A to A end to end going to the red panda but the red panda was never connected to the first harness box with power.

briskspirit, 2022-10-26:
Ok, RP looks visually ok.
Checked your logs - looks like there is a problem with your bus 0 through which OP tries to command steering. It is in bus off state. 
Have you tried to flip harness boxes? 
Also would help to see two short routes with rlogs: first normal one and second where you flip side of USB-C going from harness to red panda.

jdtram, 2022-10-26:
8a53aa70d88313f1|2022-10-25--18-26-37--0  
8a53aa70d88313f1|2022-10-25--18-29-52--0 - USB C OTG Comma side rotated 180 degrees

jdtram, 2022-10-26:
Also, yes I've switched between 3 harness boxes. Pulled one from my Elantra.

briskspirit, 2022-10-26:
No, I meant usb-c cable that connects red panda to harness box

jdtram, 2022-10-26:
Oh! Let me try that. Sorry.

jdtram, 2022-10-26:
8a53aa70d88313f1|2022-10-25--18-39-35--0 with both ends of the USB C cable flipped.
8a53aa70d88313f1|2022-10-25--18-40-46--0 with USB C cable swapped end for end. 

I've also tried a different USB C cable. I had an L shaped one that came from the car I traded in.

briskspirit, 2022-10-26:
plz flip only one side of the cable, not both. just to confirm. still see bus 0 issue.
car doesn't fault?

jdtram, 2022-10-26:
With one side flipped I get the CAN Bus Disconnected likely cable fault error from above.

jdtram, 2022-10-26:
8a53aa70d88313f1|2022-10-25--19-08-41--0 with one side flipped and can bus error.  No errors on car dash.

jdtram, 2022-10-26:
[image]
I'm not sure what this part is called but that's the only part I've not been able to swap aside from the Red Panda. I have a wiring harness connector ordered to see if the harness is the problem, hopefully that part comes with the $100 harness and not the $200 package.

briskspirit, 2022-10-26:
Connect your device to WiFi, I need to get rlogs

jdtram, 2022-10-26:
They're uploading now. Sorry, didn't realize it was offline.

briskspirit, 2022-10-26:
You will get return label tomorrow from support, suspect it's a hardware problem with red panda, need to see it.

jdtram, 2022-10-26:
Ah that's unfortunate we're driving cross country on Friday. Is there any way I can buy a red panda and have it shipped out faster than usual?

jdtram, 2022-10-26:
Disregard that question. I ordered another Red Panda and had it shipped to our destination. Hopefully customer service can process a refund for one of them.

closed 2022-10-26 by adeebshihadeh