commaai/openpilot #26235: feature: one-press speed setting during NDOG

samcooler, 2022-10-25:
When NDOG is enabled and the gas pedal is used to override the speed above the current set speed, the cruise control "set" button could be pressed once to set the speed at the current value.

Example use case: Entering highway with OP enabled, currently set to e.g. 35 mph for surface streets. Press gas to bring car up to highway speed, merging into traffic as necessary, where it is still useful to have OP lat enabled. Now at 65 mph, there is no simple way to smoothly fully engage OP, so one has to press and hold "speed up" cruise button and wait for set speed to reach current speed, or perform a rapid cancel/set two-button press, with associated lateral car jerk. The proposed feature: A single "set" button press during NDOG should match the current speed and enable release of the gas pedal, which would be much smoother here and in a variety of situations. This feature does not conflict with the other use of the button at that time, since there is little use to "set speed down" during NDOG above the current set speed.

sunnyhaibin, 2022-10-26:
It already does this in the latest `master`: https://github.com/commaai/openpilot/commit/dcde942d9fb05575dcc70f0b24f9c3c6b7c220ab

sshane, 2022-10-26:
openpilot already does this when when pcmCruise is false. If it's not, the car controls the set speed and we don't have control over that