commaai/openpilot #26262: No barcode two pair a new device

Jameselectronis, 2022-10-27:
### Describe the bug

Hi everyone I'm having problem with my comma , too is not showing barcode in order to pair the device. Any help or advice would be much appreciated
[image]

### Provide a route where the issue occurs

When trying to install

### openpilot version

0.8.13.1

### Additional info

Hi everyone I'm having problem with my comma , too is not showing barcode in order to pair the device. Any help or advice would be much appreciated
[image]

Jameselectronis, 2022-10-27:
[image]
Here is the whole thing

vanillagorillaa, 2022-10-30:
Devices that are already paired do not show the QR code. This device is either paired to your own account already, or if you bought it used, paired to the previous owners account.

vanillagorillaa, 2022-10-30:
Reading the linked PR, you purchased this from Ebay, which means this device is likely still paired to previous owner. You will need to get in touch with them and have them unpair it from their account.

closed 2022-10-30 by vanillagorillaa