commaai/openpilot #26288: Subaru D harness

adeebshihadeh, 2022-10-30:
I've seen this mentioned on Discord. Which cars need this new D harness?

kevinhofmaenner, 2023-02-04:
Hello, I have a 2023 Subaru Outback and will be purchasing a comma 3 once its supported. Is there anything I can do to move this forward? 

I created this schematic from the 2023 Outback a while ago. I can also provide more pictures if needed.

I am willing to be a beta tester for the new harness as well.

Thanks!

[image]

martinl, 2023-02-24:
Here is the 2023 Outback Eyesight Rf21 pinout from STIS
[image]

```
01 - Br - (H)_A/S CAN_L
02 - L  - P/U1 CAN_L
03 
04
05 - L  - HRLY
06
07 - Be - CRG
08 - B  - GND
09 - W  - IG
10 - Sb - +B
11 - B  - GND
12 - R - HTHR
13 - G  - (H)_A/S CAN_H
14 - R  - P/U1 CAN_H
15
16
17
18
19 - Gr - CR+
20 - Y/B - CRM
21 - W  - IG
22 - B  - GND
23 - Sb - +B
24 - L  - THRG
```

2023 WRX is using same pinout with a different wire colors
```
01 - Be - (H)_A/S CAN_L
02 - L  - P/U1 CAN_L
03 
04
05 - Sb - HRLY
06
07 - P  - CRG
08 - B  - GND
09 - R  - IG
10 - W  - +B
11 - B  - GND
12 - Gr - HTHR
13 - G  - (H)_A/S CAN_H
14 - R  - P/U1 CAN_H
15
16
17
18
19 - Y  - CR+
20 - V  - CRM
21 - R  - IG
22 - B  - GND
23 - W  - +B
24 - L  - THRG
```

vanillagorillaa, 2023-03-14:
@martinl , Ideally we can get a more clear name for each of the wires from Subaru techinfo. I understand GND is ground. Is "IG" Ignition? What is the 12 volt power source, +B? Which CAN is the powertrain CAN that we want to tap into, P/U1 CAN? We really don't care much about the other signals as they are all passthrough and can put them wherever on the comma harness. If you don't have a current techinfo sub that you can easily look this up, we can do it here.

ilyk, 2023-03-14:
I have a '23 Subaru Ascent Touring, and I had an opportunity to build the Subaru D harness out of the A one. 
Parts list:
 - 1318917-1 (Male socket plug)
 - 1376103-1 (Female socket plug)
 - few of those 1123343-1 (Female cable sockets)
 - few of those 1376109-1 (Male cable pins)

Both original (car and camera) plugs were depinned, and original wires were used to pin two 24p sockets corresponding to the scheme provided by @martinl. Actually, it was 1-to-1 as the [(Subaru-C scheme)](https://github.com/commaai/neo/blob/master/car_harness/Subaru_C_Harness.sch.pdf) with few differences:
 - instead of using 4 and 16 pins for CAN_1, I put blue and pink wires, two pins upper, into 2 and 14 correspondingly;
 - pin_4 from `1318917-1` was looped directly to the pin_16 of `1376103-1`
 - pin_16 from `1318917-1` was looped directly to the pin_4 of `1376103-1`

This DIY harness allowed me to connect the Comma Three to my '23 Subaru Ascent. The Openpilot was able to detect all needed ECUs (some issues that aren't related to the harness topic), and the original functionality of the auto wasn't broken and continues to work if the Comma Three is either turned off or disconnected, which told us that all loops are connected correctly.

martinl, 2023-03-14:
> @martinl , Ideally we can get a more clear name for each of the wires from Subaru techinfo. I understand GND is ground. Is "IG" Ignition? What is the 12 volt power source, +B? Which CAN is the [...]

Yes, IG is ignition and B+ is 12V. A/S CAN has EPS and P/U1 has powertrain related ecus (ECM, TCM), same as GEN2
https://github.com/martinl/openpilot/wiki/CAN-Network-diagrams#2020my-outbacklegacy-can-diagram

I saved the 2023 models Eyesight wiring diagrams as pdf-s but I do not have a active techinfo subscription to check if there is more info.

vanillagorillaa, 2023-04-03:
@ilyk are you on our community discord? Could you message me directly there? I have the same username.

AlterTheEnding, 2023-04-15:
Is there any news on the Subaru d harness or if another harness worked, I saw some talk of Subaru c possibly working. I just bought a 23 Outback and desperately need a harness for my C3

AlterTheEnding, 2023-04-16:
> Hello, I have a 2023 Subaru Outback and will be purchasing a comma 3 once its supported. Is there anything I can do to move this forward?
>
>
>
> I created this schematic from the 2023 Outback a while [...]

Any news on the harness?

AlterTheEnding, 2023-05-04:
@ilyk do you have any Subaru D harnesses for sale? I desperately need one.

ilyk, 2023-05-04:
@AlterTheEnding nope. I made only one for my self

AlterTheEnding, 2023-05-09:
> @AlterTheEnding nope. I made only one for my self

If I follow your diagram above will it all work? You still using yours no issues? I have someone making me a D harness from A, hopefully using your diagram/info above.

martinl, 2023-05-12:
> @martinl , Ideally we can get a more clear name for each of the wires from Subaru techinfo. I understand GND is ground. Is "IG" Ignition? What is the 12 volt power source, +B? Which CAN is the [...]

I found the techinfo Rf21 connector pinout descriptions and updated the list above

delaware-dave, 2023-05-14:
I too have purchased an 2023 Outback, Touring XT. Extremely interested in the new harness. Following

james-fourth, 2023-05-18:
@ilyk So, if someone were going to build themselves a D harness, would you suggest they get an A or C as a starting point? If the C connector matches that of the 2023 models, wouldn't it be simpler to rewire a C harness?

vanillagorillaa, 2023-09-21:
On the shop! Closing! Pusing to neo shortly

closed 2023-09-21 by vanillagorillaa