commaai/openpilot #26307: Some bugs in use[version:dragonpilot v2022.09.07,git branch:DP817C2,git commit:e091c9161e]

asd121500, 2022-11-01:
### Describe the bug

Hello, thank you very much for your efforts to let me experience the fun of automatic driving. I bought a set of your equipment and connected it to my car. Everything works normally. I went to the road to measure it and found some problems:

1. You can't turn automatically at a turn with a large bending angle, but stop driving.

2. When the car is in the left lane, turn on the turn signal, and the car will automatically turn left. In fact, it can no longer turn left at this time. If you continue to turn left, you will hit the wall.

3. It is impossible to identify pedestrians in front during driving. When there are pedestrians in front, the car will not slow down or stop automatically.

4. When passing other cars during the driving process, I was about 1m away from other cars in the transverse direction at the beginning, and he would automatically approach other cars to make the transverse distance approach about 15cm. Is this caused by my wrong use?

In view of the above issues, I would like to ask whether there is an upgraded version that can better optimize these issues, and whether a version with more accurate recognition and better use experience will be available to users in the future. If there is a new version in the future, when will it be?

### Which car does this affect?

Toyota Leiling

### Provide a route where the issue occurs

？？

### openpilot version

[version:dragonpilot v2022.09.07,git branch:DP817C2,git commit:e091c9161e]

eFiniLan, 2022-11-01:
> 

Hi @asd121500 , thanks for your comments:

for your information, we don't sell any equipment, we only develop this for fun, so if someone told you that they owned/developed the software then it's definitely not from us.

1. it's really car dependent thing, some cars turn better, others don't, it's a hardware limitation, not software. 
2. the software is not designed for this purpose, so there is nothing we can do.
3. the software is based on (openpilot)[https://github.com/commaai/openpilot], so it's up to them to add such a feature.
4. leiling (corolla?) as far as I know it is one of the best cars to run openpilot/dragonpilot, the car should keep at least 4m away from lead car, I bet it's a car limitation which it can't stop for over 3 secs, once it's over 3 secs it release the brake?

there is better hardware (comma.ai C3) out there (around 2000 USD) and it will perform better in the future because it has wide and narrow cameras. (and you will be able to use the latest AI models from comma.ai team)

C2 is outdated and will never have any AI improvements in the future.

adeebshihadeh, 2022-11-01:
We don't look into bugs from forks; open an issue on the dragon pilot repo instead.

closed 2022-11-01 by adeebshihadeh