commaai/openpilot #26349: Cruise Control Unavailable Message/Lockout after turning and activating with Toyota Corolla Hatchback 2020

nelsonjchen, 2022-11-03:
### Describe the bug

Turned off parking lot, started to try to engage OP. Car errored out with cruise control unavailable. Thankfully it was a short drive. 

As a sidenote, but probably not useful:
Think I may have seen something similar in a past route with my previous C3 in Pittsburgh (071ba9916a1da2fa) which I have have lost the route id to in the C3 shuffling. Similar incident, a turn and tried activating. https://www.google.com/maps/search/costco/@40.4064534,-79.9143638,127a,35y,5.09h,45t/data=!3m1!1e3

### Which car does this affect?

Toyota Corolla Hatchback 2020

### Provide a route where the issue occurs

fe18f736cb0d7813|2022-11-02--16-49-56--2

### openpilot version

f11f33e90b21da9fd40a5e820d8cf069c2abac3c

sshane, 2022-11-07:
As you can see here, openpilot saw the brake falling and the cruise state rising on the same frame, while the panda likely got the cruise state message slightly before the brake release, causing a panda noEntry.

[image]

nelsonjchen, 2022-11-07:
Hmm, that does make sense. I think I've been using OP as a brake hold alternative more and more recently since it's easier to turn on than brake hold and more temporary.

nelsonjchen, 2022-11-17:
Whoops, I unpreserved the route. Anyway, let me know if there's a desire for me to try and re-produce this and preserve it if the data wasn't copied and saved yet.

sshane, 2022-12-03:
Found the CAN signal that described ACC faults. Logging here so users aren't as confused if these errors occur: https://github.com/commaai/openpilot/pull/26682

[image]