commaai/openpilot #26350: “Front camera unavailable” error

MJR1989, 2022-11-03:
### Describe the bug

Had an issue with a “front camera unavailable” error appearing on my dash in my 2021 corolla hybrid for 30 seconds to 1 minute believe the error occurs between segment 27-30 idk what happened but my comma three was still responding to me pressing button on the steering wheel although I didn’t attempt to activate openpilot as I was driving through a trailer park at slow speed didn’t restart anything and error cleared itself

### Provide a route where the issue occurs

b0a7a22dea55a1cd|2022-11-03--08-33-30--27

### openpilot version

0.8.16

sshane, 2022-11-30:
This is likely a bad cable. Can you try to replace it and report back?

sshane, 2022-12-03:
I checked and there's a possible fault alert in segment 25. I also see that LKA_STATE nor LTA_STATE reported any issues from the EPS, so steering performance was likely not degraded when you saw this alert.

[image]

sshane, 2023-01-17:
Let me know if you see this again, feel free to tag me and/or re-open @MJR1989!

closed 2023-01-17 by sshane