commaai/openpilot #26384: car docs: handle US vs international models better

adeebshihadeh, 2022-11-05:
Our required package/trim section is for the most part a US market reference, however international models are often very different. They can have different trims, packages, or even entirely different ADAS systems (e.g. Honda Bosch vs Nidec).

imranali4, 2022-11-30:
Thanks this can significantly increase your international sales.

adeebshihadeh, 2023-04-04:
Thought about this a bit more, and I think we want to document two things:
1. easiest thing to understand for US market, e.g. "all packages and variants", "Safety Package II", "Honda Sensing", etc.
2. what are the most basic things I need the car to come equipped with? this will be universally true, regardless of market
   * this will depend on the manufacturer. for some, we'll still be able to reference packages. for the rest, the packages/trims vary by market, so we'll need to list specific features, e.g. "Lane Departure Alert with Steering Assist (LDA w/SA)" instead of the "Toyota Safety Sense P" package

We have #1 documented, so we'll just need to add #2 and surface it nicely.

closed 2024-08-09 by adeebshihadeh