commaai/openpilot #26394: Ionity 2019 CAN error

danevito, 2022-11-07:
### Describe the bug

Switched from Ionity 2020 plug-in to the Ionity electric 2019 (a supported car). Haven't got it working with either car. All it says is "CAN Error: Check connections" after a few seconds of booting. Uninstalled and reinstalled stock, swapped the cables, ( I have two sets of most cables because a reorder of new car harness)  

With the Ionity 2020 I just got "non-supported car" message. 

[image]

### Provide a route where the issue occurs

26e73a0e32642dc4

### openpilot version

0.8.16

### Additional info

this bug report is for the 2019 Ionity electric.

sshane, 2022-11-07:
It appears your car isn't broadcasting the FCA (forward collision assist) message, and we couldn't get any ABS firmware either. Does this car have Smart Cruise Control (it accelerates and brakes for cars), as well as automatic emergency braking?

danevito, 2022-11-08:
wow you can read all that from the short time I was online?  are you sure you are looking at the right car and not the earlier ionig plug-in 2020? 

to answer your question it does have forward collision assist. not sure what ABS firmware is? dosent all cars today have ABS breaking?

Q: will C3 not detect if you disable a feature in the setting like forward assist?

sunnyhaibin, 2022-11-09:
> Does this car have Smart Cruise Control...

This part is vital to know from your 2019 Ioniq EV.

danevito, 2022-11-13:
Just realised, its Ioniq not Ionity ofc.

danevito, 2022-11-27:
https://i.ibb.co/9sW1xyw/20221120-022652.jpg
https://i.ibb.co/k6LmZcS/20221120-022635.jpg

When I go to the CC settings in the car, while the C3 is on I can't change the settings and it's stuck at "regular LKA" instead of "Active LKA" which I guess would be the correct settings? 

If I remove/turn off the C3 I can change the settings freely again.  

A thought: can the language difference in the car mess with the comma ai code somehow?

sshane, 2022-12-03:
Have you disabled forward collision assist?

sunnyhaibin, 2022-12-05:
@danevito can you please confirm if your car is a **2019 Hyundai Ioniq Electric**, with **Smart Cruise Control**?

It would also be very helpful if you can take a picture of all the buttons on the steering wheel in a single picture.

danevito, 2022-12-05:
[image]

[image]

danevito, 2023-01-10:
I am still wondering if the language difference can cause some confusion in C3?  

Have other people with Ioniqs reported problem after a open pilot version upgrade in the past?

danevito, 2023-01-10:
Is this related??

https://github.com/commaai/openpilot/pull/20706

sunnyhaibin, 2023-01-15:
@danevito can you post a route with this issue on the latest `master` branch, with raw logs uploaded?

> • Share your device with me (https://connect.comma.ai/, gear icon, share by email, sunnyhaibin@gmail.com, share icon again to confirm), set the device name to your year/make/model and your Discord username (with #xxxx) so I can find it in the list.
> • Go to https://connect.comma.ai, select the route with the issue, More info, next to "View in cabana" -> click "upload", then Files, View upload queue, and confirm all raw logs are uploaded.
> • Post the route ID in this channel.

danevito, 2023-01-16:
26e73a0e32642dc4|2023-01-15--12-02-50--1

danevito, 2023-01-16:
I am only using the dashcam at the moment, do you want me to upload a route with open pilot on?

sunnyhaibin, 2023-01-16:
> I am only using the dashcam at the moment, do you want me to upload a route with open pilot on?

@danevito going to need a route that has "CAN error" generated.

danevito, 2023-01-16:
26e73a0e32642dc4|2023-01-16--19-40-46--1  

this is the new one with CAN error

sshane, 2023-01-17:
Are you sure you have a 2020 Ioniq EV? I checked your FW versions and they more closely match the 2019 Ioniq EV Limited, which when I run your route under that car, everything works fine as it expects the FCA message to be bad or not exist and doesn't rely on it. Perhaps your 2020 Ioniq EV is actually the American 2019 model year, as that would be a good explanation of why your car doesn't match our expectations of a 2020

sshane, 2023-01-17:
Are you sure you have a 2020 Ioniq EV? I checked your FW versions and they more closely match the 2019 Ioniq EV Limited, which when I run your route under that car, everything works fine as it expects the FCA message to be bad or not exist and doesn't rely on it. Perhaps your 2020 Ioniq EV is actually the same as the American 2019 model year, as that would be a good explanation of why your car doesn't match our expectations of a 2020.

You were CAN fingerprinting to the 2020 because we were missing CAN fingerprints for the 2019, so once I add your versions to the 2019 this should be resolved.

@danevito can you confirm which harness letter type you purchased, and did it fit without any issues?

sshane, 2023-01-17:
Ah, found the issue, not sure why I didn't see it before. Your car is fingerprinting as the 2020 Ioniq EV for two reasons:

- We don't have your car's exact FW versions in the database
- We only have a CAN fingerprint for the 2020 Ioniq EV

I added your versions here on this branch, let me know if you get a chance to test it:

https://github.com/commaai/openpilot/pull/26988
Installer URL: https://installer.comma.ai/commaai/fp-ioniq-20-as-19

To confirm, the Hyundai C harness worked for your vehicle?

sshane, 2023-01-17:
@danevito merged, you should be good to go on `master` or `master-ci` branches:

Installer URL is now:

https://installer.comma.ai/commaai/master-ci

closed 2023-01-17 by sshane

danevito, 2023-01-18:
cheers m8, looks to be working alright.   uploaded and shared my first open pilot ride here: 26e73a0e32642dc4|2023-01-18--15-03-00--0

can you explain the difference in master-ci and master?  are both non-tested builds?  

when will I be able to update to the current latest model?