commaai/openpilot #26415: on-screen prompt when killing openpilot while on-road

gregjhogan, 2022-11-08:
If you `CTRL+C` in the tmux session and openpilot is on-road, we could add a confirmation dialog on the device (to ensure you are doing this to the intended device).

adeebshihadeh, 2022-11-08:
I don't think this is worth writing or maintaining. It's not really a problem if this ever happens; this is well within what one should expect when driving with openpilot. Also, it's largely only a problem we're exposed to internally, and it doesn't really happen.

closed 2022-11-08 by adeebshihadeh

gregjhogan, 2022-11-08:
I was thinking non-technical people do give access to more technical community members for support, so it didn't seem limited to just us, and I couldn't think of a situation where you should be killing openpilot while on-road without being in the vehicle.