commaai/openpilot #26457: Wide camera display shifts to the right

oremax, 2022-11-10:
### Describe the bug

When I tried it on 0.8.17 master-ci
The wide camera display shifts to the right.
Just driving is no problem.
It's a display issue only.

The display with the normal camera is working normally without any problems.

What caused this issue
Camera calibration → NG
UNINSTALL recovery → NG

At low speeds, it is always off, so
I want to turn off the automatic change to the wide camera.

Device used
C3

car
PRIUSPRIME2021

UI that shows roads and lanes are always misaligned
https://photos.app.goo.gl/bjokWfz7wB1ELydR6

### Provide a route where the issue occurs

516d6c3bd3ce6315|2022-11-08--07-35-14--2

### openpilot version

0.8.17 master-ci

adeebshihadeh, 2022-11-10:
All your routes are on forks. If you can reproduce on stock master-ci, feel free to re-open.

closed 2022-11-10 by adeebshihadeh

oremax, 2022-11-19:
commaai
release3-staging
Version 0.9.0 (2022-11-21)

Wherever you use the wide camera, it is always shifted to the right.

516d6c3bd3ce6315|2022-11-19--08-02-57--0
516d6c3bd3ce6315|2022-11-19--08-02-57--0

https://youtu.be/YGnTI0NCjIA

Is it a hardware problem?