commaai/openpilot #26458: setup issue

jeroenlammersma, 2022-11-10:
### Describe the bug

Currently, you are trying to `source` the current SHELL `rc` in the [update_requirements.sh](https://github.com/commaai/openpilot/blob/e08896c45d84f4005de156317ae902d601e9b895/update_requirements.sh) script.

Example:
```bash
RC_FILE="${HOME}/.$(basename ${SHELL})rc"
source $RC_FILE
```

a.k.a.
```bash
source ~/.bashrc
```

However, it is not possible to do this sort of action: [see explanation](https://askubuntu.com/questions/64387/cannot-successfully-source-bashrc-from-a-shell-script).

The current setup could *potentially* work for an working version of an openpilot dev environment. However, for greenfield users, running the setup for the first time, this results in an error:

``` bash
WARNING: seems you still have not added 'pyenv' to the load path.

# Load pyenv automatically by appending
# the following to
~/.bash_profile if it exists, otherwise ~/.profile (for login shells)
and ~/.bashrc (for interactive shells) :

export PYENV_ROOT="$HOME/.pyenv"
command -v pyenv >/dev/null || export PATH="$PYENV_ROOT/bin:$PATH"
eval "$(pyenv init -)"

# Restart your shell for the changes to take effect.

# Load pyenv-virtualenv automatically by adding
# the following to ~/.bashrc:

eval "$(pyenv virtualenv-init -)"

pyenv update ...
/home/jeroen/openpilot/update_requirements.sh: line 35: pyenv: command not found
```

### OS Version

Ubuntu 20.04

jeroenlammersma, 2022-11-10:
Also..

If you execute the ```ubuntu_setup.sh``` for a second time, this error will not occur. You might think the issue is **now** fixed, however...

```bash
jeroen@phoenix:~/openpilot$ poetry shell
Spawning shell within /home/jeroen/.cache/pypoetry/virtualenvs/openpilot-eYl9iItR-py3.8
jeroen@phoenix:~/openpilot$ . /home/jeroen/.cache/pypoetry/virtualenvs/openpilot-eYl9iItR-py3.8/bin/activate
(openpilot-py3.8) jeroen@phoenix:~/openpilot$ scons -u -j$(nproc) # executing scons as advertised..
scons: Reading SConscript files ...

scons: warning: Calling missing SConscript without error is deprecated. # whoops! Error...
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'cereal/SConscript'
File "/home/jeroen/openpilot/SConstruct", line 357, in <module>

scons: warning: Ignoring missing SConscript 'rednose/SConscript'
File "/home/jeroen/openpilot/SConstruct", line 394, in <module>

scons: warning: Ignoring missing SConscript 'cereal/SConscript'
File "/home/jeroen/openpilot/SConstruct", line 408, in <module>

scons: warning: Ignoring missing SConscript 'body/board/SConscript'
File "/home/jeroen/openpilot/SConstruct", line 408, in <module>

scons: warning: Ignoring missing SConscript 'panda/board/SConscript'
File "/home/jeroen/openpilot/SConstruct", line 408, in <module>

scons: warning: Ignoring missing SConscript 'opendbc/can/SConscript'
File "/home/jeroen/openpilot/SConstruct", line 408, in <module>

scons: *** Import of non-existent variable ''libkf''
File "/home/jeroen/openpilot/selfdrive/locationd/SConscript", line 1, in <module>
(openpilot-py3.8) jeroen@phoenix:~/openpilot$
```

edgarwideman, 2022-11-25:
Wow, I wish I checked for issues here 12 hours ago. I literally TORE my hair out all day trying to figure out why I am getting this error. I am a Linux noob (have no idea what I'm doing) trying to get open pilot running in Ubuntu on [Framework](https://frame.work) laptop. If someone could let me know how to resolve this issue let me know, I would be thrilled.

PR #26601 (2022-11-30): tools: pyenv setup fix
resolves #26458 by immediately setting up environment for pyenv

also removes requirement for installing sudo
closed 2022-11-30 by incognitojam