commaai/openpilot #26483: Routes don't seem to upload

korhojoa, 2022-11-12:
### Describe the bug

Drives won't, well, upload. 

I tried to make this route upload a few times, clicking the appropriate buttons in connect.comma.ai but they didn't seem to want to leave the device.

The route does exist on the device still, but as the time for me to be able to see it has passed, it's a bit hard to test.

```
comma@tici:/data/openpilot$ ls -la /data/media/0/realdata/2022-11-06--19-00-19--79/
total 190500
drwxr-xr-x   2 comma comma     4096 Nov  6 20:20 .
drwxr-xr-x 584 comma comma    36864 Nov 12 09:23 ..
-rw-r--r--   1 comma comma 75097388 Nov  6 20:20 ecamera.hevc
-rw-r--r--   1 comma comma 75126526 Nov  6 20:20 fcamera.hevc
-rw-r--r--   1 comma comma  2232312 Nov  6 20:20 qcamera.ts
-rw-r--r--   1 comma comma  2626584 Nov  6 20:20 qlog
-rw-r--r--   1 comma comma 39915632 Nov  6 20:20 rlog
```

### Provide a route where the issue occurs

a954cbd0682c58bd|2022-11-06--19-00-19--79

### openpilot version

647ec1e56770429c17bf82cc7cd155dcca84aebd

### Additional info

Bandwidth should be ok, device is connected via ethernet, speedtest looks like this:

```
   Speedtest by Ookla

      Server: RETN - Helsinki (id: 31122)
         ISP: Elisa
Idle Latency:     1.47 ms   (jitter: 0.08ms, low: 1.43ms, high: 1.60ms)
    Download:   243.98 Mbps (data used: 239.7 MB)
                 83.50 ms   (jitter: 59.22ms, low: 1.80ms, high: 832.27ms)
      Upload:   214.10 Mbps (data used: 104.2 MB)
                  7.15 ms   (jitter: 1.45ms, low: 3.63ms, high: 14.61ms)
 Packet Loss:     0.0%
 ```

adeebshihadeh, 2022-11-15:
@incognitojam can you check this out?

incognitojam, 2022-11-15:
I can see the route finished uploading to storage on Nov 6th, 10:34 AM PST (93 segments) and about 40 rlogs were uploaded on Nov 7th by 10:14 PM PST.

incognitojam, 2022-11-23:
Closing this since I couldn't find any error messages or problems with the device. Please open another issue if you can reproduce this!

closed 2022-11-23 by incognitojam