commaai/openpilot #26488: modeld/onnxmodel.cc Assertion failed

franco126, 2022-11-13:
### Describe the bug

Hi !
I'm trying to use Openpilot with Carla Simulator on PC. I follow this steps:

- run Carla (not the one provided by openpilot in docker) 
-- `./CarlaUE4.sh -opengl -nosound -RenderOffScreen -benchmark -fps=20 -quality-level=Low`
- `./launch_openpilot.sh` (poetry shell)
- `./bridge.py` (poetry shell)

However, I got this error and **camerad** and **modeld** are not working

`Onnx selected provider:  ['CUDAExecutionProvider']
Traceback (most recent call last):
  File "/home/francesco/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 72, in <module>
    ort_session = ort.InferenceSession(sys.argv[1], options, providers=[provider])
  File "/home/francesco/.cache/pypoetry/virtualenvs/openpilot-t1CjC73B-py3.8/lib/python3.8/site-packages/onnxruntime/capi/onnxruntime_inference_collection.py", line 347, in __init__
    self._create_inference_session(providers, provider_options, disabled_optimizers)
  File "/home/francesco/.cache/pypoetry/virtualenvs/openpilot-t1CjC73B-py3.8/lib/python3.8/site-packages/onnxruntime/capi/onnxruntime_inference_collection.py", line 384, in _create_inference_session
    sess = C.InferenceSession(session_options, self._model_path, True, self._read_config_from_model)
onnxruntime.capi.onnxruntime_pybind11_state.InvalidProtobuf: [ONNXRuntimeError] : 7 : INVALID_PROTOBUF : Load model from models/supercombo.onnx failed:Protobuf parsing failed.
_modeld: selfdrive/modeld/runners/onnxmodel.cc:64: void ONNXModel::pwrite(float *, int): Assertion err >= 0 failed.
`

If I play with bridge.py I get this message from Openpilot: "**Openpilot unavailable, press set to engage**"
[image]
[image]

### OS Version

Ubuntu 20.04

jbeck9, 2022-12-01:
Hey there,

Your driving model probably wasn't downloaded when you cloned the repo. Go look at `models/supercombo.onnx` and see. If your model file is <1kb... it ain't your driving model. Re-download from the repo and replace, and that should fix it.

adeebshihadeh, 2022-12-01:
@jbeck9 is right. make sure to do a `git lfs pull`

closed 2022-12-01 by adeebshihadeh

zeroyang0809, 2023-02-12:
i meet the same problem. Do you fix it? could you show some steps to fix it? Thanks

jbeck9, 2023-02-12:
> i meet the same problem. Do you fix it? could you show some steps to fix it? Thanks

1. Make sure git-lfs is installed. https://github.com/git-lfs/git-lfs/blob/main/INSTALLING.md

2. `git lfs pull <openpilot url>`

Note that this can only happen if you are building the OP docker from source. If you aren't modifying OP, you can also just use the docker image that is provided `  docker pull ghcr.io/commaai/openpilot-sim:latest`