commaai/openpilot #26503: Carla simulation not available in Openpilot UI

franco126, 2022-11-15:
### Describe the bug

When I run the simulation on PC, Carla "interface" in Openpilot UI is not shown. 
So I can't see car or road.
[image]
When I press 1 in bridge.py shell, it says "Openpilot available press set to engage. 
In launch_openpilot.sh shell seems to be everything ok.
Even if I run in docker container, I get the same issue
Here some screenshot...
[image]
[image]

### OS Version

Ubuntu 20.04

### Additional info

### Steps to reproduce

- Terminal 1: run CARLA (not docker version)
- Terminal 2: poetry shell - cd /tools/sim - ./launch_openpilot.sh
- Terminal 3: poetry shell - cd tools/sim - ./bridge.py

Ruishenl, 2022-11-24:
I ran into the exact same issue. I am on Ubuntu22

eli-davis, 2022-11-27:
same issue, ubuntu 20, followed all directions...doesnt show up on ui

jbeck9, 2022-12-01:
Same. Openpilot seems to run normally, just nothing on the UI. This is running Carla and OpenPilot through the Docker images for me.

I'm guessing it is related to #26169, but that is just a guess.

didpurwanto, 2022-12-06:
Same here. Carla is working fine but the bridge in 0.9.0 version has nothing.

joeking11829, 2022-12-07:
Same issue here.

fatsheep2020, 2022-12-11:
same issue here

didpurwanto, 2022-12-12:
Any updates about this?
In my case rn, I can see the future trajectory, but cannot load the images from Carla.

ebadi, 2022-12-13:
> Same. Openpilot seems to run normally, just nothing on the UI. This is running Carla and OpenPilot through the Docker images for me.

> same issue, ubuntu 20, followed all directions...doesnt show up on ui

@jbeck9 and @eli-davis, how do you know that openpilot is working?

For me OP seems not to be **fully** engaged after I press `1`. 
[image]

ebadi, 2022-12-13:
Passing ` --high_quality --dual_camera` to `bridge.py` solved the camera issue for me. The vehicle is still not moving with my interference using `WASD` .

[image]

didpurwanto, 2022-12-14:
@ebadi try press 1, then press 2 (twice), then click multiple times at the screen. In my case, I will get the trajectory by doing this.

Also me, after rebuild again. I got like yours too.
[image]

didpurwanto, 2022-12-14:
@ebadi I also got this too. But, there is no future trajectory.

didpurwanto, 2022-12-14:
@ebadi same here.
[image]

Ruishenl, 2022-12-14:
I tried with the latest Docker Image again on master branch, but the UI  now doesn't  even change to the running mode, it stays at the home page. Is there a way we can rebuild the docker image on our own ?

didpurwanto, 2022-12-15:
Check opencv2, mine is missing
ImportError: libnppc.so.11: cannot open shared object file: No such file or directory
@ebadi

ebadi, 2022-12-15:
[didpurwanto](https://github.com/didpurwanto) Please check this bug report and workaround (reinstalling `opencv-python-headless`) here #26786

zeroyang0809, 2023-02-20:
hello i meet the problem "openpilot unavailable waiting for controls to start"
this problem is familiar with your problem
do you know how to fix it ?
thx

adeebshihadeh, 2023-03-14:
Fixed with d5375af2f5c9f74cb88ffa1ab52feb9fef77a43c

closed 2023-03-14 by adeebshihadeh