commaai/openpilot #26564: [‘19 Kia Sorento] Large difference in cluster and C3 speeds

kalub92, 2022-11-21:
### Describe the bug

The speed on my dash doesn’t match the speed on my C3 by sometimes up to 25+ MPH. In the photo I attached you can see the cluster says 68 MPH but the C3 says 42 MPH.

### Which car does this affect?

2019 Kia Sorento

### Provide a route where the issue occurs

b78a19b557b991ab|2022-11-20--12-38-48--2

### openpilot version

0.9.0

### Additional info

[image]

adeebshihadeh, 2022-11-22:
@sshane Did we miss this in our analysis? Is it just this individual car or the whole model?

sshane, 2022-11-22:
@kalub92 can you put your device on WiFi so it can upload the full logs? Thanks

sshane, 2022-11-22:
@kalub92 did you wipe your device recently? No rlogs are found on the device

kalub92, 2022-11-22:
> @kalub92 did you wipe your device recently? No rlogs are found on the device for even the latest route

Ah yes, sorry I switched branches to `devel` to toy with experimental OP long. Will make sure rlogs are set to upload and update this soon.

sshane, 2022-11-22:
Checked a bunch of segments and it looks like your car is the only one in our dataset that sets `CF_Clu_VehicleSpeed` with an imperial value, which we assume is always metric/kph. Will need to see if it's this platform or your car, as we only had your dongle for the Sorento in the data

sshane, 2022-11-22:
@kalub92 can you try this branch? https://github.com/commaai/openpilot/pull/26582

Installer URL: https://installer.comma.ai/commaai/sorento-match-dash

Can you also try to test both imperial (mph) and metric (kph)? You'll find the setting somewhere on your center console or dash.

kalub92, 2022-11-23:
> @kalub92 can you try this branch? https://github.com/commaai/openpilot/pull/26582
> 
> 
> 
> Installer URL: https://installer.comma.ai/commaai/sorento-match-dash
> 
> 
> 
> Can you also try to test both imperial (mph) and metric (kph)? You'll find the setting somewhere on your car's center console or dash.

Just tried and confirmed that it works for imperial: `b78a19b557b991ab|2022-11-22--21-42-08--0`.

However, I wasn't able to switch my cars speed unit. I searched around online and wasn't able to find much for a '19 Sorento.

kalub92, 2022-11-23:
> Nice! It's a bit buried in your head unit's vehicle settings, but it should be there. Let me know if you try to find the setting again or not

Well the weird thing is I went into Settings > Other and see temperature unit, fuel unit, and one other unrelated setting, but nothing about speed unit. I’ll take another look tomorrow!

kalub92, 2022-11-24:
> Nice! It's a bit buried in your head unit's vehicle settings, but it should be there. Let me know if you try to find the setting again or not

I looked through every menu and submenu in my head unit's settings and I searched through the menus in Engineering mode through the cars stereo to no avail. I've found a few other people expressing the same issue with this particular year of car: https://www.kia-forums.com/threads/trying-to-change-speedometer-from-km-hr-to-mph-on-2019-sorento-sxl-canada.344253

sshane, 2022-12-06:
I also checked the other two Sorento dongles (that are only ever on forks or a C2), and this signal is the same on all of them. One is American/imperial and the signal is already in mph (like user above). The other is metric (the unit value is 0), and the signal is in metric already (as we expect). So conclusion is, if there's another car where `CF_Gway_DrLockSoundRValue` is set to 3, and they have this mismatch, we can safety use that.

sshane, 2022-12-06:
@kalub92 merged a fix, let me know if you have any other issues!

closed 2022-12-06 by sshane