commaai/openpilot #26577: Process not running - loggerd, rawgpsd

Balirb, 2022-11-22:
### Describe the bug

Version 0.9.0, C3, Honda Odyssey 2018, Device-ID: 22159f038ee65583

openpilot not engaging since the update (from 0.8.16 to 0.9.0). Following error is triggered when I try to engage openpilot: Process not running - loggerd, rawgpsd

### Provide a route where the issue occurs

22159f038ee65583

### openpilot version

0.9.0

### Additional info

Now this error is stopping me to engage openpilot. However I was able to engage couple times on 11/21 after restarting the device - restarting is not working anymore to bypass this error.

WillDeStijl, 2022-12-23:
Similar error:

 Process not running
encoderd

----
Toyota Corolla 2019 ; Comma Three ; 0.9.0-release / release3 / Nov 22

WillDeStijl, 2022-12-23:
Reboot "solved" it.

Will report back if recurring

kile-lindgren, 2023-01-14:
I'm getting a similar error. Process not running rawgpsd.

Version 0.9.0, C3, 2023 Toyota Highlander

Erich helped get the fingerprints updated to recognize my car.

Route: 5b6b8032ce973622|2023-01-14--00-44-52

If I unplug the  usbc cable, start the car, and plug it back in. I have had luck getting openpilot to engage.  But after I turn off the car and start it. I get the same rawgpsd not started error.  Rebooting the comma 3 when the error occurs has no effect. Pulling power cable is the only thing that works (so far).

adeebshihadeh, 2023-01-14:
> I'm getting a similar error. Process not running rawgpsd.
>
> Version 0.9.0, C3, 2023 Toyota Highlander
>
> Erich helped get the fingerprints updated to recognize my car.
>
> Route: [...]

This is a different issue and possibly a malfunction of the modem. Email support@comma.ai.

M3cklar, 2023-01-15:
I’m getting the same error on my comma 3 with 0.9.0. https://photos.app.goo.gl/jvH1wapseVG8NVNCA

adeebshihadeh, 2023-02-03:
Ah, I see you sent your device in for a bad NVMe drive.

closed 2023-02-03 by adeebshihadeh

jonlove234, 2023-02-03:
fwiw, I've had this error and have an SSD that is otherwise functioning. a total restart of car & fully-draining c3 capacitors springs thing into live, but obviously it's not an ideal solution in the middle of a drive.

M3cklar, 2023-02-03:
I can’t be sure it was the ssd itself or something else. I did not get any details from Comma on what the issue was, but they are sending me a warranty replacement.

travelswagger, 2023-02-12:
I continue to encounter occasional process not available loggerd, rawgpsd, gpxd. Connect is not showing any routes on these drives even though C3 status shows online and all good. I don't know how else to share documentation or evidence of this bug.

M3cklar, 2023-02-12:
it took me a while to get a warranty replacement from Comma. They required
I run stock OP and also send them routes, but having the same issue as you
I had no routes to send. They eventually just sent me a new one after I
returned my problem unit.

My new unit loaded fine up right away on stock OP and routes were
immediately visible. My new device now has a problem with fan malfunction
and overheating on Stock. Trying to get that figured out.

travelswagger, 2023-02-12:
Thank you. I just sent them an email. I decided to wait until a new stock Openpilot master was released for my Bolt EUV, and it seems to have fixed the long-standing LKAS error. 

Unfortunately, loggerd, rawgpsd, gpxd all fail to run more frequently compared to successful error-free drives. I've refreshed Agnos, installed master-ci and master. I've reseated all of the USB-C connections at the harness (flipped too) and unit. I guess we'll see what happens with my 1TB Crosscountry and hope your fan issue is resolved.

adeebshihadeh, 2023-02-12:
> My new unit loaded fine up right away on stock OP and routes were immediately visible. My new device now has a problem with fan malfunction and overheating on Stock. Trying to get that figured out.

I just checked out your support ticket; support will get back to you soon to have it sent back. Looks like your fan stopped working after the first few drives, which we haven't really seen. The fan connector may have just come unplugged.

> Thank you. I just sent them an email. I decided to wait until a new stock Openpilot master was released for my Bolt EUV, and it seems to have fixed the long-standing LKAS error.

Support will get back to you soon about those process not running errors; those particular ones are generally indicative of hardware failures. As for the Bolt faults, we expect those to be fixed on master and the upcoming 0.9.1 release.

jonlove234, 2023-03-04:
[image]

Still getting this

adeebshihadeh, 2023-03-04:
We can't look into issues from forks; report them to your fork maintainer instead.