commaai/openpilot #26595: Carla Openpilot Not Working with the latest docker image

Ruishenl, 2022-11-24:
### Describe the bug

I am following the instruction to get the carla-openpilot integration working, but the dashboard doesn't seem to render the carla world. I am able to pass my keyboard input to engage openpilot but got Openpilot Unavailable message:
[image]

Steps to reproduce:
Terminal 1
./start_carla.sh

Terminal 2 
./start_openpilot_docker.sh

Terminal 3:
docker exec -it OPENPILOT_CONTAINER_ID /bin/bash
./launch_openpilot.sh

Terminal 4
docker exec -it OPENPILOT_CONTAINER_ID /bin/bash
./bridge.py

### OS Version

Pop!_OS 22.04

### openpilot version or commit

b15018387c774b9f470e88f22b96861d62917656

### Additional info

I also have a warning message showing the current branch is not tested. Am i supposed to use a released tag instead ?

fatsheep2020, 2022-12-11:
@incognitojam  do it simulator issue? anddo you know how to fix it, as i knew, it work well before....
thanks a lot!

incognitojam, 2022-12-16:
This is the same issue as #26503

closed 2022-12-16 by incognitojam

zeroyang0809, 2023-02-16:
hello do you fix this problem? could you show solutions to fix it? 
thanks