commaai/openpilot #26602: Looking for fingerprinting or car support

AlterTheEnding, 2022-11-26:
### Describe the bug

I recently got the comma three and my device will only work in dashcam mode on my 22 Subaru Forester Wilderness. I’m new to all this so I hope I report all this correctly, I apologize in advance. 

### Which car does this affect?

Subaru Forester Wilderness 2022

### Provide a route where the issue occurs

8db3f557a1e70a1a|2022-11-26--19-49-23

### openpilot version

Master-ci

### Additional info

I first tried the open pilot release dashcam mode only then the Subaru communities fork/branch (sorry new here) and that was dash cam mode only then read the help page for fingerprinting and tried master-ci version and still dashcam mode only. I truly appreciate your help!

AlterTheEnding, 2022-11-26:
Thank you for labeling all this correctly for me, I appreciate your help on this.

martinl, 2022-11-27:
I looked at the route and required carFw values for Subaru are missing Eyesight ecu response (0x787).

We also tried to run fw_versions.py --scan on c3 via ssh but no luck:
```
Found FW versions
{
  Brand: hyundai   , bus: 1 - (Ecu.fwdRadar, 0x7d0, None): [b'\xf1\x00\x00\x00\x02']
  Brand: hyundai   , bus: 1 - (Ecu.fwdCamera, 0x7c4, None): [b'\xf1\x00\xac\x02\x00']
  Brand: hyundai   , bus: 1 - (Ecu.transmission, 0x7e1, None): [b'\xf1\x00\xd7\x10@']
  Brand: hyundai   , bus: 1 - (Ecu.fwdRadar, 0x7d0, None): [b'\xf1\x00\x00\x00\x02']
  Brand: hyundai   , bus: 1 - (Ecu.transmission, 0x7e1, None): [b'\xf1\x00\xd7\x10@']
  Brand: subaru    , bus: 1 - (Ecu.abs, 0x7b0, None): [b'\xa3 !x\x00']
  Brand: subaru    , bus: 1 - (Ecu.eps, 0x746, None): [b'-\xc0%0']
  Brand: subaru    , bus: 1 - (Ecu.engine, 0x7e0, None): [b'\xd5"a0\x07']
  Brand: subaru    , bus: 1 - (Ecu.transmission, 0x7e1, None): [b'\x1d\x86B0\x00']
  Brand: subaru    , bus: 1 - (Ecu.unknown, 0x717, None): [b'\xb3 !\x01\x00']
  Brand: volkswagen, bus: 1 - (Ecu.transmission, 0x7e1, None): [b'\xf1\x82\x1d\x86B0\x00']
}

Possible matches: set()
Getting fw took 4.573 s
```

then stopped openpilot and ran ./panda/examples/query_fw_versions.py
```
comma@tici:/data/openpilot$ ./panda/examples/query_fw_versions.py
opening device 0b0026001850335332313620 0xddcc
connected
querying addresses ...
0x18dafff1: 100%|████████| 512/512 [01:49<00:00,  4.69it/s]

 Results for address 0x717 

0xF182 APPLICATION_DATA_IDENTIFICATION: b'\xb3 !\x01\x00'
0xF186 ACTIVE_DIAGNOSTIC_SESSION: b'\x03'
0xF197 SYSTEM_NAME_OR_ENGINE_TYPE: b'Electric Brake Booster          '
0xF198 REPAIR_SHOP_CODE_OR_TESTER_SERIAL_NUMBER: b'\x11\x11\x11\x11'

 Results for address 0x742 

0xF182 APPLICATION_DATA_IDENTIFICATION: b'\xf1\x00\x01'
0xF186 ACTIVE_DIAGNOSTIC_SESSION: b'\x03'
0xF189 VEHICLE_MANUFACTURER_ECU_SOFTWARE_VERSION_NUMBER: b'050000\x00\x00\x00\x00'
0xF18E VEHICLE_MANUFACTURER_KIT_ASSEMBLY_PART_NUMBER: b'87611VC002'
0xF197 SYSTEM_NAME_OR_ENGINE_TYPE: b'RADAR ASSY B&S LH               '
0xF198 REPAIR_SHOP_CODE_OR_TESTER_SERIAL_NUMBER: b'\xff\x00\x00\x00'

 Results for address 0x743 

0xF182 APPLICATION_DATA_IDENTIFICATION: b'\xf1\x10\x01'
0xF186 ACTIVE_DIAGNOSTIC_SESSION: b'\x03'
0xF189 VEHICLE_MANUFACTURER_ECU_SOFTWARE_VERSION_NUMBER: b'050000\x00\x00\x00\x00'
0xF18E VEHICLE_MANUFACTURER_KIT_ASSEMBLY_PART_NUMBER: b'87611VC002'
0xF197 SYSTEM_NAME_OR_ENGINE_TYPE: b'RADAR ASSY B&S RH               '
0xF198 REPAIR_SHOP_CODE_OR_TESTER_SERIAL_NUMBER: b'\xff\x00\x00\x00'

 Results for address 0x746 

0xF182 APPLICATION_DATA_IDENTIFICATION: b'-\xc0%0'
0xF197 SYSTEM_NAME_OR_ENGINE_TYPE: b'Power Steering System           '

 Results for address 0x747 

0xF182 APPLICATION_DATA_IDENTIFICATION: b'!2$\x10'
0xF186 ACTIVE_DIAGNOSTIC_SESSION: b'\x03'
0xF189 VEHICLE_MANUFACTURER_ECU_SOFTWARE_VERSION_NUMBER: b'6046P-10'
0xF18E VEHICLE_MANUFACTURER_KIT_ASSEMBLY_PART_NUMBER: b'84051SJ060'
0xF197 SYSTEM_NAME_OR_ENGINE_TYPE: b'Automatic Headlight Leveling    '
 Results for address 0x752 *

[... 92 lines cut ...]

0xF182 APPLICATION_DATA_IDENTIFICATION: b'\x02\x9f\x81\x02\x00\x04\x84\x82\x10\x00'
0xF186 ACTIVE_DIAGNOSTIC_SESSION: b'\x03'
0xF189 VEHICLE_MANUFACTURER_ECU_SOFTWARE_VERSION_NUMBER: b'4848100020'
0xF18E VEHICLE_MANUFACTURER_KIT_ASSEMBLY_PART_NUMBER: b'85261SJ810'
0xF197 SYSTEM_NAME_OR_ENGINE_TYPE: b'MFD                             '
0xF198 REPAIR_SHOP_CODE_OR_TESTER_SERIAL_NUMBER: b'\x7f$\x9dX'

 Results for address 0x7E0 

0xF182 APPLICATION_DATA_IDENTIFICATION: b'\xd5"a0\x07'
0xF186 ACTIVE_DIAGNOSTIC_SESSION: b'\x03'
0xF187 VEHICLE_MANUFACTURER_SPARE_PART_NUMBER: b'\x00\x00\x00\x00\x00\x00\x00\x00'
0xF190 VIN: b'JF2xxxxxxxxxxxxxxxxxx'
0xF197 SYSTEM_NAME_OR_ENGINE_TYPE: b'2.5 DOHC                        '

 Results for address 0x7E1 

0xF182 APPLICATION_DATA_IDENTIFICATION: b'\x1d\x86B0\x00'
0xF186 ACTIVE_DIAGNOSTIC_SESSION: b'\x03'
0xF197 SYSTEM_NAME_OR_ENGINE_TYPE: b'CVT                             '
comma@tici:/data/openpilot$
```
so Eyesight is still 0x787 and fw version is `b'\x04!\x01\x1eD\x07!\x00\x04,'`
could this be timing issue?

martinl, 2022-11-28:
Also, 2022 Forester does not have CruiseControl On and Activated signals, same as 2020+ Forester Hybrid. We'll test with ES_DashStatus
[image]

martinl, 2022-11-28:
ES_LKAS is also not used and there is new message 0x124
[image]

sshane, 2022-11-28:
Looks like 0x787 responds negatively to the tester present query we make, so it doesn't continue to get the FW version.

martinl, 2022-12-04:
[image]

mwbarker, 2024-08-19:
So did this work? Thinking about getting a comma for my 2024 Forester, but don’t want to spend that much if it is only a dashcam. Glad to help where I can with development, though (testing, etc.).