commaai/openpilot #26605: C3 crashing and getting into bad state

mogorman, 2022-11-28:
### Describe the bug

My C3 is sometimes getting into a bad state, refusing to boot, beeps of death and rebooting, and getting into a bad state where its frozen or controls are not available.  I have had this issue with both jvepilot fork based on 0.8.15 and stock openpilot 0.9  Support asked me to report the bug here, I personally believe this is a hardware issue but wanted to move this forward.  The route provided is one where it died and the processes zombied.  the additional info is what i could get from the device before rebooting it.

### Provide a route where the issue occurs

ea9cb83a8559a88f|2022-11-26--21-15-57--2

### openpilot version

0.9.0

### Additional info

$ ps aux | grep 'Z'

```
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
comma      46035  0.0  0.0      0     0 pts/1    Z+   Nov26   0:18 [_modeld] <defunct>
comma      46043  0.0  0.0      0     0 pts/1    Z+   Nov26   0:13 [selfdrive.locat] <defunct>
comma      46058  0.0  0.0      0     0 pts/1    Z+   Nov26   0:13 [selfdrive.monit] <defunct>
comma      46060  0.0  0.0      0     0 pts/1    Z+   Nov26   0:25 [selfdrive.locat] <defunct>
comma      46064  0.0  0.0      0     0 pts/1    Z+   Nov26   0:01 [selfdrive.senso] <defunct>
comma      46065  0.0  0.0      0     0 pts/1    Z+   Nov26   0:01 [selfdrive.navd.] <defunct>
comma      46068  0.0  0.0      0     0 pts/1    Z+   Nov26   0:26 [selfdrive.locat] <defunct>
comma      46071  0.0  0.0      0     0 pts/1    Z+   Nov26   0:18 [selfdrive.senso] <defunct>
comma      46075  0.0  0.0      0     0 pts/1    Z+   Nov26   0:46 [selfdrive.contr] <defunct>
```
tmux is filled with this message
`shutting device down, offroad since 59499.384098149`

and dmesg has
```
[126154.788415] R0: [wpa_supplicant][2422323620948] [13:17:42.377516]  wlan: [4539:W :SME] sme_ft_update_key: 359: Unhandled state: 0
[126154.788456] R0: [cds_mc_thread][2422323621800] [13:17:42.377560]  wlan: [45662:I :WMI] send_per_roam_config_cmd_tlv: 12424: per roam enable=3, vdev=0
[126154.788685] R0: [cds_mc_thread][2422323626169] [13:17:42.377788]  wlan: [45662:I :WMI] send_roam_scan_offload_rssi_change_cmd_tlv: 12485: roam_scan_rssi_change_thresh=5, bcn_rssi_weight=14
[126154.788710] R0: [cds_mc_thread][2422323626494] [13:17:42.377805]  wlan: [45662:I :WMI] send_roam_scan_offload_rssi_change_cmd_tlv: 12486: hirssi_delay_btw_scans=15000
[126154.788779] R0: [cds_mc_thread][2422323628010] [13:17:42.377884]  wlan: [45662:I :WMI] WMI --> WMI_ROAM_AP_PROFILE and other parameters
[126154.788828] R0: [cds_mc_thread][2422323628948] [13:17:42.377932]  wlan: [45662:I :WMI] 2452,
[126154.788839] R0: [cds_mc_thread][2422323629158] [13:17:42.377943]  wlan: [45662:I :WMI] 2432,
[126154.788850] R0: [cds_mc_thread][2422323629366] [13:17:42.377954]  wlan: [45662:I :WMI] 5200,
[126154.788860] R0: [cds_mc_thread][2422323629567] [13:17:42.377965]  wlan: [45662:I :WMI] 5745,
[126154.788871] R0: [cds_mc_thread][2422323629773] [13:17:42.377975]  wlan: [45662:I :WMI] 5240,
[126154.788888] R0: [cds_mc_thread][2422323629978] [13:17:42.377986]  wlan: [45662:I :WMI] 5180,
[126154.788899] R0: [cds_mc_thread][2422323630320] [13:17:42.378004]  wlan: [45662:I :WMI] 5805,
[126154.788910] R0: [cds_mc_thread][2422323630527] [13:17:42.378015]  wlan: [45662:I :WMI] 2422,
[126154.788921] R0: [cds_mc_thread][2422323630726] [13:17:42.378025]  wlan: [45662:I :WMI] 5765,
[126154.789007] R0: [cds_mc_thread][2422323632378] [13:17:42.378111]  wlan: [45662:I :WMI] LFR3:OKC enabled
[126154.789018] R0: [cds_mc_thread][2422323632599] [13:17:42.378123]  wlan: [45662:I :WMI] LFR3:PMKSA caching enabled
[126154.789431] R0: [cds_mc_thread][2422323640509] [13:17:42.378535]  wlan: [45662:I :SME] ########### Set Key Complete #############
[126154.789516] R0: [cds_mc_thread][2422323642130] [13:17:42.378619]  wlan: [45662:E :WMA] Unhandled WMA message of type 4209
[126154.793171] R0: [systemd-network][2422323712245] [13:17:42.382271]  wlan: [4418:E :HDD] hdd_is_rcpi_applicable: 4948: mac addr is different from bssid connected
[126154.814929] R0: [networkctl][2422324129934] [13:17:42.404025]  wlan: [116507:E :HDD] hdd_is_rcpi_applicable: 4948: mac addr is different from bssid connected
[126154.831171] R0: [iwconfig][2422324441769] [13:17:42.420266]  wlan: [116508:W :HDD] __hdd_ioctl: 7734: unknown ioctl 35587
[126154.833336] R0: [iwconfig][2422324483401] [13:17:42.422434]  wlan: [116508:W :HDD] __hdd_ioctl: 7734: unknown ioctl 35593
[126156.258445] R0: [cds_mc_thread][2422351845440] [13:17:43.847518]  wlan: [45662:E :WMA] wma_log_completion_timeout: Timeout occured for log completion command
[126171.829455] IRQ237 no longer affine to CPU5
[126171.829464] IRQ238 no longer affine to CPU5
[126171.829471] IRQ239 no longer affine to CPU5
[126171.829476] IRQ240 no longer affine to CPU5
[126171.829483] IRQ241 no longer affine to CPU5
[126171.829489] IRQ242 no longer affine to CPU5
[126171.829496] IRQ243 no longer affine to CPU5
[126171.829502] IRQ244 no longer affine to CPU5
[126171.829508] IRQ245 no longer affine to CPU5
[126171.829592] IRQ565 no longer affine to CPU5
[126296.721147] R0: [cds_mc_thread][2425048729077] [13:20:04.307911]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 867: Beacon for session[0] got changed.
[126296.721209] R0: [cds_mc_thread][2425048730357] [13:20:04.307978]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 869: sending beacon param change bitmap: 0x10
[126296.721317] R0: [cds_mc_thread][2425048732411] [13:20:04.308085]  wlan: [45662:E :PE ] sch_process_pre_beacon_ind: 1015: PreBeaconInd received in invalid state: 11
[126306.551113] R0: [cds_mc_thread][2425237463462] [13:20:14.137666]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 867: Beacon for session[0] got changed.
[... 227 lines cut ...]
[126376.314369] send_afe_cal_type cal_block not found!!
[126376.323273] q6asm_send_cal: cal_block is NULL
[126376.354524] event_handler: reclaimed all bufs
[126376.383462] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4098
[126376.383499] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4098
[126376.383525] send_afe_cal_type cal_block not found!!
[126376.392917] q6asm_send_cal: cal_block is NULL
[126376.411369] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4101
[126376.411411] afe_get_cal_topology_id: [AFE_TOPOLOGY_CAL] not initialized for this port 4101
[126376.411444] send_afe_cal_type cal_block not found!!
[126376.411472] send_afe_cal_type cal_block not found!!
[126376.420050] q6asm_send_cal: cal_block is NULL
[126476.631360] R0: [cds_mc_thread][2428503005185] [13:23:04.215133]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 867: Beacon for session[0] got changed.
[126476.631416] R0: [cds_mc_thread][2428503006341] [13:23:04.215193]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 869: sending beacon param change bitmap: 0x10
[126476.631513] R0: [cds_mc_thread][2428503008206] [13:23:04.215290]  wlan: [45662:E :PE ] sch_process_pre_beacon_ind: 1015: PreBeaconInd received in invalid state: 11
[126486.781296] R0: [cds_mc_thread][2428697883660] [13:23:14.364884]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 867: Beacon for session[0] got changed.
[126486.781390] R0: [cds_mc_thread][2428697885558] [13:23:14.364982]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 869: sending beacon param change bitmap: 0x10
[126486.781558] R0: [cds_mc_thread][2428697888811] [13:23:14.365152]  wlan: [45662:E :PE ] sch_process_pre_beacon_ind: 1015: PreBeaconInd received in invalid state: 11
[126496.816516] R0: [cds_mc_thread][2428890559973] [13:23:24.399940]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 867: Beacon for session[0] got changed.
[126496.816604] R0: [cds_mc_thread][2428890561702] [13:23:24.400030]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 869: sending beacon param change bitmap: 0x10
[126496.816766] R0: [cds_mc_thread][2428890564828] [13:23:24.400193]  wlan: [45662:E :PE ] sch_process_pre_beacon_ind: 1015: PreBeaconInd received in invalid state: 11
[126507.261752] R0: [cds_mc_thread][2429091108430] [13:23:34.844998]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 867: Beacon for session[0] got changed.
[126507.261843] R0: [cds_mc_thread][2429091110274] [13:23:34.845094]  wlan: [45662:W :PE ] __sch_beacon_process_for_session: 869: sending beacon param change bitmap: 0x10
[126507.262012] R0: [cds_mc_thread][2429091113530] [13:23:34.845263]  wlan: [45662:E :PE ] sch_process_pre_beacon_ind: 1015: PreBeaconInd received in invalid state: 11
```

mogorman, 2022-11-30:
in my dmesg i am also seeing some issues with the camera?
```
Aug 31 11:27:33 tici kernel: CAM_ERR: CAM-IRQ-CTRL: cam_irq_controller_subscribe_irq: 268 Invalid params: bh_handler=ffffff84cd35fe58 b>
Aug 31 11:27:33 tici kernel: CAM_ERR: CAM-ISP: cam_vfe_bus_deinit_hw: 2933 Failed to unsubscribe error irq rc=-22
Aug 31 11:27:33 tici kernel: CAM_ERR: CAM-IRQ-CTRL: cam_irq_controller_subscribe_irq: 268 Invalid params: bh_handler=ffffff84cd35fe58 b>
Aug 31 11:27:33 tici kernel: CAM_ERR: CAM-ISP: cam_vfe_bus_deinit_hw: 2933 Failed to unsubscribe error irq rc=-22
Aug 31 11:27:33 tici kernel: CAM_ERR: CAM-UTIL: cam_soc_util_get_option_clk_by_name: 536 No clk named ife_dsp_clk found. Dev vfe2
Aug 31 11:27:33 tici kernel: CAM_WARN: CAM-ISP: cam_vfe_init_soc_resources: 107 option clk get failed
Aug 31 11:27:33 tici kernel: CAM_ERR: CAM-IRQ-CTRL: cam_irq_controller_subscribe_irq: 268 Invalid params: bh_handler=ffffff84cd35fe58 b>
Aug 31 11:27:33 tici kernel: CAM_ERR: CAM-ISP: cam_vfe_bus_deinit_hw: 2933 Failed to unsubscribe error irq rc=-22
Aug 31 11:27:33 tici kernel: CAM_INFO: CAM-ISP: cam_isp_dev_probe: 172 Camera ISP probe complete
Aug 31 11:27:33 tici kernel: CAM_INFO: CAM: cam_res_mgr_probe: 704 Enable shared gpio support.
Aug 31 11:27:33 tici kernel: CAM_ERR: CAM-SMMU: cam_smmu_create_add_handle_in_table: 689 Error: cam-secure already got handle 0x4c4d8
Aug 31 11:27:33 tici kernel: CAM_INFO: CAM-JPEG: cam_jpeg_dev_probe: 170 Camera JPEG probe complete
```

mogorman, 2022-12-03:
Another crashed route ea9cb83a8559a88f|2022-12-03--11-28-54--0

adeebshihadeh, 2022-12-07:
This is a case of #24302.

closed 2022-12-07 by adeebshihadeh

mogorman, 2022-12-07:
removing my drive for time being to see if it makes a difference. will report back.

mogorman, 2022-12-31:
Zero problems since I removed the nvme drive from the c3