commaai/openpilot #26612: Completely reliable fuzzy fingerprinting

adeebshihadeh, 2022-11-28:
A user with a supported car should never have to deal with a fingerprint. We'll need two things for this: 100% reliable fuzzy fingerprinting and an automated path for fuzzy fingerprints to make their way into openpilot. The FW versions checked into openpilot will remain as a list of verified, known working ECUs.

I imagine we can do this in a pretty safe, but brand-specific way. For example, Hyundai FW versions seem to all be prefixed with the car model.

adeebshihadeh, 2022-11-28:
#24757 has an approach for cars that get ECU updates

adeebshihadeh, 2022-11-29:
@sshane let's get this shipped for Toyota or Hyundai first, whichever is easier. then other contributors can take a stab at it for their cars

MoreTore, 2023-04-11:
Use the Dcam to recognize the interior of the car 🤣

MoreTore, 2023-04-11:
What if there was a new panel in settings with a list of supported cars? If the car is unrecognized, the user should be able to select from a list of supported vehicles. If the user selects a vehicle and goes for a drive, and the drive does not have any bad events, the fw_versions can be written to a file and used next time.

closed 2023-06-15 by adeebshihadeh