commaai/openpilot #26624: Can never enable Open Pilot, Cruise Fault: Restart the car to engage, caused by low_speed_lockout event

jieyuwuCar, 2022-11-29:
### Describe the bug

When running stock 8.14 on my 2017 prius two, I did not have any issues.  But when updating to 0.9 I basically can no longer run stock 0.9 at this point.  Every time I start the car i get the low_speed_lockout event.  I'd have to disable low_speed_lockout event by hard coding that line away in order to just run stock normally. 

If I comment these lines out from /openpilot/selfdrive/car/toyota/interface.py:
    if self.CS.low_speed_lockout and self.CP.openpilotLongitudinalControl:
      events.add(EventName.lowSpeedLockout)

Then I can engage Openpilot again when driving over 30mph.

### Which car does this affect?

Toyota Prius Two 2017

### Provide a route where the issue occurs

Occurs on startup

### openpilot version

0.9

jieyuwuCar, 2022-11-29:
[image]

nelsonjchen, 2022-11-30:
You should include your dongle id if you haven't already so comma can look at the logs.

incognitojam, 2022-12-02:
@jieyuwuCar Please share a link to a route where this happens or the route name (find in https://connect.comma.ai or https://useradmin.comma.ai)

jieyuwuCar, 2022-12-05:
@nelsonjchen I had to downgrade it from 0.9 to 8.14 because I am doing a lot of trips lately.  So looking at the logs won't be helpful at the moment.

@incognitojam It's not a route that it happens.  This happens when Comma boots up.  I can never engage open pilot due to that low_speed_lockout event.

incognitojam, 2022-12-05:
A route starts when you are onroad (you can see the camera), so even though you are not engaged there are still logs created. What is your dongle ID?

sshane, 2022-12-06:
0.8.14 lets you engage without any changes, but 0.9 does not @jieyuwuCar?

sshane, 2023-08-08:
Unfortunately KWP2000 doesn't have a service to control communication, we can only put it in programming mode to silence the output. @jieyuwuCar does this still happen with DSU unplugged with the recent onroad transition speed up (setting power save mode)?

jieyuwuCar, 2023-08-08:
I bought a SDSU and the SDSU fixed the issue.

closed 2024-04-17 by sshane