commaai/openpilot #26657: Car left lane in curve

MJR1989, 2022-12-01:
### Describe the bug

Car crossed halfway into next lane at approximately 35 mph in a curve

### Provide a route where the issue occurs

b0a7a22dea55a1cd|2022-12-01--10-19-51--7

### openpilot version

0.9.0

incognitojam, 2022-12-02:
Looks like the actual lateral accel overshot what openpilot wanted here for about 2.5 seconds
[image]

MJR1989, 2022-12-03:
> Looks like the actual lateral accel overshot what openpilot wanted here for about 3-4 seconds
> 
> [image]
> 
> 
> 
> 

I've had a lot of issues since 0.9.0 came out with oversteer normally it's not so dramatic normally it's either the inside tire riding the line like a rail at slower speeds or coming within inches of the lane line at highway speeds idk what causes this hopefully someone way smarter than me can fix it 😅

sshane, 2023-08-08:
@nuwandavek have you looked into this? Friction too high?

adeebshihadeh, 2023-12-17:
Thanks for the report. Please post future driving feedback in the #driving-feedback channel on Discord.

closed 2023-12-17 by adeebshihadeh