commaai/openpilot #26677: Lateral over-correcting in longer/harder corners, riding close or on top of line

foodbandlt, 2022-12-02:
### Describe the bug

There seems to be some sort of overcorrection on about 50% of the corners that I take that lead to the car riding very close to, or  perfectly on top of the lane line as seen in the sideview mirror.  This is a little concerning for 2-lane roads where there may be another car oncoming.

**Nuwandavek** in Discord looked into this with this response, but I believe they've gotten too busy to look into this further: 
> I looked into a lot of your routes. I can see that the learner works as expected, but the steering torque - lateral acceleration relationship is slightly non linear at the extremes. This is very different from all the other Toyotas I've seen. I'll have a branch for you to try out shortly, with some fixes.

Routes uploaded and preserved: 
b4f83330b2d59205|2022-11-17--17-14-31--6
b4f83330b2d59205|2022-11-17--13-34-15--4 through segment 8
b4f83330b2d59205|2022-11-17--12-58-17--2
b4f83330b2d59205|2022-11-17--09-16-54--3 (was told torque values weren't learned yet)

### Which car does this affect?

Toyota Camry Hybrid 2022

### Provide a route where the issue occurs

b4f83330b2d59205|2022-11-17--17-14-31--6

### openpilot version

0.9

foodbandlt, 2022-12-02:
I will say that this did not happen the the older NG model, this is a new behavior for me personally.

morrislee, 2022-12-02:
this started happening to me from rocket launcher model (original NG performed a lot better)

I am on a taller platform, not sure if it is the same issue, Harold thinks its difference in height, be camery is pretty small compared to my Sierra 1500AT4

foodbandlt, 2023-02-02:
Two more particularly noticeable segments on the new WD model on master:

b4f83330b2d59205|2023-01-30--19-12-30--7
b4f83330b2d59205|2023-01-30--19-12-30--10

It seems to only be low speeds, highway speeds are literally perfect for cornering, even the sharper ones.   I also made a thead in #driving-feedback in the discord server, feel free to close if that's more appropriate than an issue now that that's up and running!

sshane, 2023-03-25:
Looks like classic turn cutting in the last two examples you sent, the car was matching the requested turning acceleration exactly. Is this still happening on master with the new 2023-driving model merged in?

closed 2023-08-07 by YassineYousfi